stlink-org/stlink #574: stink v1 driver issues on macOS Sierra

karpenet, 2017-03-18:
Unable to build driver on macOS Sierra 10.12.3

Output:

```
OS X version not supported.
make: *** [osx_stlink_shield] Error 1
```

Expected/description:
`Tried building with Yosemite and El Capitan configurations as in driver readme, none seem to work.`

xor-gate, 2017-03-18:
I'm not sure you need the driver for stlinkv1 on newer OS X. If you get your hands on a stlinkv2 this is recommended as they also can measure target voltage. The stlinkv1 is deprecated some time ago by ST Microelectronics.

karpenet, 2017-03-18:
A lot of Discovery boards still ship with the stink-v1. I'm using the STM32VLDiscovery board which has this version too.

xor-gate, 2017-03-18:
Have you tried texane/stlink tools without driver?

amok, 2017-04-02:
@xor-gate I tried - it does't work.
ST-LINK V2 works, but STM32VLDiscovery with its v1 - doesn't
it's even unable to find a device

```
➜  system_profiler SPUSBDataType
USB:

    USB 3.0 Bus:

      Host Controller Driver: AppleUSBXHCIWPT
      PCI Device ID: 0x9cb1
      PCI Revision ID: 0x0003
      PCI Vendor ID: 0x8086

         ...

        STM32 STLink:

          Product ID: 0x3744
          Vendor ID: 0x0483  (STMicroelectronics)
          Version: 1.00
          Serial Number: Vÿk?RpQPV"?g
          Speed: Up to 12 Mb/sec
          Manufacturer: STMicroelectronics
          Location ID: 0x14100000 / 9
          Current Available (mA): 500
          Extra Operating Current (mA): 0

➜  st-util
st-util 1.3.1
2017-04-02T18:36:39 WARN src/usb.c: Couldn't find any ST-Link/V2 devices

➜  lsusb
...
Bus 020 Device 009: ID 0483:3744 STMicroelectronics STM32 STLink  Serial: Vÿk?RpQPV"?g

➜  STLINK_DEVICE=020:009 st-util
st-util 1.3.1
2017-04-02T18:38:33 INFO src/usb.c: bus 020 dev 009
2017-04-02T18:38:33 WARN src/usb.c: Couldn't find matched ST-Link/V2 devices

➜  st-util -1
st-util 1.3.1
2017-04-02T18:40:55 WARN src/sg.c: Failed to find an stlink v1 by VID:PID
2017-04-02T18:40:55 ERROR src/sg.c: Could not open stlink device
```

xor-gate, 2017-04-03:
Probably the kernel attaches to SCSI driver, so you are unable to see it.

amok, 2017-04-03:
ioreg says

```
+-o STM32 STLink@14100000  <class AppleUSBDevice, id 0x100007d84, registered, matched, active, busy 0 (28 ms), retain 16>
```
I am not sure, but shouldn't it be IOUSBAttached**SCSI** or something?

amok, 2017-04-03:
@xor-gate seems st-util is able to detect and connect to board using prior version of libusb, I tried v1.0.9

```
➜  stlink git:(master) ./build/Release/src/gdbserver/st-util
st-util 1.3.1-12-g6902f47
2017-04-04T02:04:39 INFO src/common.c: Loading device parameters....
2017-04-04T02:04:39 INFO src/common.c: Device connected is: F1 Medium/Low-density Value Line device, id 0x10016420
2017-04-04T02:04:39 INFO src/common.c: SRAM size: 0x2000 bytes (8 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 1024 bytes
2017-04-04T02:04:39 INFO src/gdbserver/gdb-server.c: Chip ID is 00000420, Core ID is  1ba01477.
2017-04-04T02:04:39 INFO src/gdbserver/gdb-server.c: Listening at *:4242...
```

libusb@master devices list
```
➜  libusb git:(master) ./examples/testlibusb
Dev (bus 20, device 5): Broadcom Corp. - Bluetooth USB Host Controller
```

libusb@v1.0.9 devices list

```
➜  libusb git:(f98eaca) ✗ ./examples/testlibusb
Dev (bus 20, device 0): 05AC - 8007
Dev (bus 20, device 5): Broadcom Corp. - Bluetooth USB Host Controller
Dev (bus 20, device 10): STMicroelectronics - STM32 STLink

➜  libusb git:(f98eaca) ✗ ./examples/listdevs
05ac:8007 (bus 20, device 0)
05ac:8290 (bus 20, device 5)
0483:3744 (bus 20, device 10)
```

xor-gate, 2017-04-04:
@amok that was a detailed investigation. I'm glad we see this fixed in libusb. Good work!

karpenet, 2017-04-10:
Thanks @amok but I'm still getting an error.

No other programs using the port.

```
$ system_profiler SPUSBDataType
USB:

    USB 3.0 Bus:

      Host Controller Driver: AppleUSBXHCIWPT
      PCI Device ID: 0x9cb1 
      PCI Revision ID: 0x0003 
      PCI Vendor ID: 0x8086 

        Bluetooth USB Host Controller:

          Product ID: 0x8290
          Vendor ID: 0x05ac  (Apple Inc.)
          Version: 1.46
          Speed: Up to 12 Mb/sec
          Manufacturer: Broadcom Corp.
          Location ID: 0x14300000 / 4
          Current Available (mA): 500
          Current Required (mA): 0
          Extra Operating Current (mA): 0
          Built-In: Yes

        STM32 STLink:

          Product ID: 0x3744
          Vendor ID: 0x0483  (STMicroelectronics)
          Version: 1.00
          Serial Number: Vÿl?QRP"1?
          Speed: Up to 12 Mb/sec
          Manufacturer: STMicroelectronics
          Location ID: 0x14200000 / 8
          Current Available (mA): 500
          Current Required (mA): 100
          Extra Operating Current (mA): 0
```

```$st-util

st-util 1.3.1-12-g6902f47
2017-04-10T14:42:11 WARN src/usb.c: Stlink usb device found, but unable to claim (probably already in use?)
[!] send_recv send request failed: LIBUSB_ERROR_NOT_FOUND
[!] send_recv STLINK_DEBUG_RESETSYS
2017-04-10T14:42:11 INFO src/gdbserver/gdb-server.c: Chip ID is 00000000, Core ID is  00000000.
2017-04-10T14:42:11 INFO src/gdbserver/gdb-server.c: Listening at *:4242...```

DouglasPearless, 2017-04-10:
I am getting exactly the same error :-(

amok, 2017-04-10:
@xor-gate  sorry for confusion
in my case last version of libusb does not work, but v1.0.9 works
so seems this bug (or feature) with devices list was introduced in some of the last versions
created issue libusb/libusb#290

@karpenet I am not sure we have the same issues (if so - sorry for spamming your thread). have you tried to downgrade libusb?

xor-gate, 2017-04-10:
Hi @amok its not a problem to discus here which probably could cause the problem.

C44Supra, 2017-04-18:
I'm having a similar issue (as per issue #587). I have built stlink from the latest source and installed it with homebrew on separate occasions. To get the Without the driver / kext I get:
```
$ st-util -1
st-util 1.3.1
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
2017-04-18T15:21:43 WARN src/sg.c: libusb_claim_interface() failed
2017-04-18T15:21:43 ERROR src/sg.c: Could not open stlink device
```

And with the driver I get:
```
$ st-util -1
st-util 1.3.1
2017-04-18T15:04:32 WARN src/sg.c: Failed to find an stlink v1 by VID:PID
2017-04-18T15:04:32 ERROR src/sg.c: Could not open stlink device
```

I've had to modify the install.sh file so that it installs the 10.10 kext on Sierra. I also have the kext-dev-mode flag set but I'm not quite sure that method even does anything anymore on Sierra. Sierra itself does detect the device (I can see it in the System Profiler, deviceID and productID seem to be correct too).

As an experiment I installed Ubuntu 17.04 in a VM on the same system, built stlink from the same source, reconnected the board to the system and when VMWare Fusion asked me to choose a "destination" for the STLink I chose "Ubuntu". Started st-util -1 and it worked flawlessly in Ubuntu.

I'm inclined to believe that this might be a kext issue. As long as there's no kext installed it is detected but can't be claimed, most likely due to the mass storage issue. Once the kext is installed however stlink can't find it.

As of this moment I'm running whatever version libusb was installed automagically (most likely 1.0.21?) and I honestly have no idea how to install an older version (like 1.0.9) to see if this resolves the problem on my end as well.

amok, 2017-04-19:
> ... how to install an older version ...

the simplest way probably would be to git clone libusb repo and then compile it on v1.0.9 tag and then symlink binaries to `/usr/local/lib/`
@C44Supra  could you please try the same - curious will it work for you

C44Supra, 2017-04-19:
@amok I've been trying to do that but I fear something is just not going quite right. 

I've got one folder in my home dir where I keep all my STM32 related stuff which is where I execute all of these commands from. Here are the literal steps that I've taken so far:

````
$ git clone https://github.com/libusb/libusb.git
$ cd libusb
$ git reset --hard ab9cd5a7be637f7b793987971a706b1d11c27ded
````
So far, so good. I've checked the files and I see numerous references to v 1.0.9. Now it get's sketchy. The first command to run, according to the INSTALL file, is './configure'. However that won't fly:
````
-bash: ./configure: No such file or directory
````
The only executable file I can see in the folder is autogen.sh, which gives me this:
````
glibtoolize: putting auxiliary files in '.'.
glibtoolize: copying file './ltmain.sh'
glibtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
glibtoolize: copying file 'm4/libtool.m4'
glibtoolize: copying file 'm4/ltoptions.m4'
glibtoolize: copying file 'm4/ltsugar.m4'
glibtoolize: copying file 'm4/ltversion.m4'
glibtoolize: copying file 'm4/lt~obsolete.m4'
configure.ac:38: installing './compile'
configure.ac:39: installing './config.guess'
configure.ac:39: installing './config.sub'
configure.ac:29: installing './install-sh'
configure.ac:29: installing './missing'
examples/Makefile.am:1: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
examples/Makefile.am: installing './depcomp'
libusb/Makefile.am:4: warning: source file 'os/darwin_usb.c' is in a subdirectory,
libusb/Makefile.am:4: but option 'subdir-objects' is disabled
automake: warning: possible forward-incompatibility.
automake: At least a source file is in a subdirectory, but the 'subdir-objects'
automake: automake option hasn't been enabled.  For now, the corresponding output
automake: object file(s) will be placed in the top-level directory.  However,
automake: this behaviour will change in future Automake versions: they will
automake: unconditionally cause object files to be placed in the same subdirectory
automake: of the corresponding sources.
automake: You are advised to start using 'subdir-objects' option throughout your
automake: project, to avoid future incompatibilities.
libusb/Makefile.am:3: warning: source file 'os/linux_usbfs.c' is in a subdirectory,
libusb/Makefile.am:3: but option 'subdir-objects' is disabled
libusb/Makefile.am:5: warning: source file 'os/openbsd_usb.c' is in a subdirectory,
libusb/Makefile.am:5: but option 'subdir-objects' is disabled
libusb/Makefile.am:6: warning: source file 'os/poll_windows.c' is in a subdirectory,
libusb/Makefile.am:6: but option 'subdir-objects' is disabled
libusb/Makefile.am:6: warning: source file 'os/windows_usb.c' is in a subdirectory,
libusb/Makefile.am:6: but option 'subdir-objects' is disabled
libusb/Makefile.am:37: warning: source file 'os/threads_windows.c' is in a subdirectory,
libusb/Makefile.am:37: but option 'subdir-objects' is disabled
libusb/Makefile.am:37: warning: source file 'os/threads_posix.c' is in a subdirectory,
libusb/Makefile.am:37: but option 'subdir-objects' is disabled
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... ./install-sh -c -d
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
[... 380 lines cut ...]

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
test -z "/usr/local/include/libusb-1.0" || .././install-sh -c -d "/usr/local/include/libusb-1.0"
 /usr/bin/install -c -m 644 libusb.h '/usr/local/include/libusb-1.0'
Making install in doc
make[2]: Nothing to be done for `install-exec-am'.
make[2]: Nothing to be done for `install-data-am'.
make[2]: Nothing to be done for `install-exec-am'.
test -z "/usr/local/lib/pkgconfig" || ./install-sh -c -d "/usr/local/lib/pkgconfig"
 /usr/bin/install -c -m 644 libusb-1.0.pc '/usr/local/lib/pkgconfig'
````

Again, I see 4 new files in /usr/local/lib, which have the exact same name as the ones that showed up using the github source.

st-util -1 gives me the same error yet again:
````
$ st-util -1
st-util 1.3.1-15-g3d24377
2017-04-20T01:45:49 WARN src/sg.c: Failed to find an stlink v1 by VID:PID
2017-04-20T01:45:49 ERROR src/sg.c: Could not open stlink device
````

Am I doing something oddly wrong here?

Nandox7, 2018-04-09:
Got the same issue and managed to sort it loading the .kext file.
I'm running 10.13.2 and just copy the 10.10 version & changed the OS version detection code as mentioned in the README.

Problem was is the new protections in 10.13.
Was forced to boot into recovery mode and disable kext signing protection using.
csrutil disable
csrutil enable --without kext

After restart was able to load the .kext and st-util -1 works correctly.

ryba-xek, 2018-07-27:
Ok, I'm still failing with OS X 10.13.6.

I've done `csrutil disable/enable --without kext` in recovery mode, booted back to normal.

I took 10.10 kext and moved it to `/System/Library/Extensions` and chown'ed it.
`> ll /System/Library/Extensions/ | grep -i stli
drwxr-xr-x  3 root  wheel    96B Jul 26 19:58 stlink_shield.kext`

When I load it I see everything's okay:
`> sudo kextload -v /System/Library/Extensions/stlink_shield.kext
Requesting load of /System/Library/Extensions/stlink_shield.kext.
/System/Library/Extensions/stlink_shield.kext loaded successfully (or already loaded).`

but kextstat gives me nothing.

`> sudo kextstat | grep -i stl
>`

st-util -1 does not work:
`> st-util -1
st-util 1.4.0
2018-07-27T17:15:39 WARN sg.c: Failed to find an stlink v1 by VID:PID
2018-07-27T17:15:39 ERROR sg.c: Could not open stlink device`

Still, kextunload works well, it reports that the kext was unloaded.
`> sudo kextunload -v /System/Library/Extensions/stlink_shield.kext
com.libusb.stlink_shield unloaded and personalities removed.`

xor-gate, 2018-07-28:
What does st-util say when you run as root with sudo? It is not a the perfect situation but we can see if it is a permission problem.

ryba-xek, 2018-07-28:
`> sudo st-util -1`
`st-util 1.4.0`
`2018-07-29T00:00:08 WARN sg.c: Failed to find an stlink v1 by VID:PID`
`2018-07-29T00:00:08 ERROR sg.c: Could not open stlink device`

I'm able to see
`STM32 STLink:

  Product ID:	0x3744
  Vendor ID:	0x0483  (STMicroelectronics)`
in Sytem Information.

This is what I see from libusb 1.0.9
`> ./examples/listdevs`
`05ac:8007 (bus 20, device 0)`
`0483:3744 (bus 20, device 8)  --- this disappers when I disconnect STM32`

The same command shows nothing in libusb-1.0.22 =(

AlfredHunter, 2018-09-06:
Ok, I'm seems OK with OS X 10.13.6.

I've done `csrutil disable` in recovery mode, booted back to normal.

I took 10.10 kext and moved it to `/System/Library/Extensions`.

When I load it I see everything's okay:
`$ sudo kextload -v /System/Library/Extensions/stlink_shield10_10.kext`
```
Requesting load of /System/Library/Extensions/stlink_shield10_10.kext.
/System/Library/Extensions/stlink_shield10_10.kext loaded successfully (or already loaded).
```
And then
`$ sudo touch /System/Library/Extensions`

And then got this below:
`$ ./build/Release/src/gdbserver/st-util -1`
```st-util 1.4.0-47-gae717b9
2018-09-06T15:22:32 WARN sg.c: Your stlink got into a real weird configuration, trying to fix it!
2018-09-06T15:22:32 INFO common.c: Loading device parameters....
2018-09-06T15:22:32 INFO common.c: Device connected is: F1 High-density device, id 0x10036414
2018-09-06T15:22:32 INFO common.c: SRAM size: 0x10000 bytes (64 KiB), Flash: 0x80000 bytes (512 KiB) in pages of 2048 bytes
2018-09-06T15:22:32 INFO sg.c: Successfully opened a stlink v1 debugger
2018-09-06T15:22:32 INFO gdb-server.c: Chip ID is 00000414, Core ID is  1ba01477.
2018-09-06T15:22:32 INFO gdb-server.c: Listening at *:4242...
```

Nightwalker-87, 2020-02-22:
Related to #852 .
As commented there we need a major update on this.
We seem to lack proper support for current macOS versions.

Nightwalker-87, 2020-04-06:
As @jasonhouang mentioned before, the root of this problem is a system security setting introduced by Apple with OS X El Capitan (OS X 10.11) in 2015.

The so called System Integrity Protection (SIP) is active per default and prevents the operating system amongst other things to load unsigned Kernel Extension Modules (kext). Thus the ST-Link-v1 driver supplied with the tools and installs as a kext, remains not functional while SIP is fully activated as per default.

Action needs to be taken here by booting into the recovery mode where a  terminal console window needs to be opened.

In contrast to the previous approach **I would NOT RECOMMEND to disable SIP completely as with the command** `csrutil disable`, **because this leaves the system more vulnerable to common threats. Instead there is a more adequate and reasonable option implemented by Apple. Running** `csrutil enable --without kext`, **allows to load unsigned kernel extensions while leaving SIP active with all other security features.
So who ever intends to run the ST-Link-v1 programmer on macOS please take this into account.**

Further details can be found here: https://forums.developer.apple.com/thread/17452

Nightwalker-87, 2020-04-06:
I am going to add the instructions above to our project documentation in order to make them publicly accessible and will subsequently close this issue as resolved.

Nightwalker-87, 2020-04-07:
Here we actually had two separate problems in one issue:
1) **libusb problems with device detection in macOS:**
Bug in external `libusb` library, _solved in v1.0.23_, which is also distributed via the package repositories  `homebrew` and `MacPorts` per default. Thus this is no longer an issue.
2) **ST-Link-v1 driver issue relates to Kernel Extensions (kext):**
Closed by commit 59c5cfd7ec5c7f2a035944ecc5ca01d3d7b31170.

commit 59c5cfd7 (2020-04-07): [doc] macOS ST-Link-v1 detection

Added documentation on how to solve a failed detection of the ST-Link-v1 
programmer in macOS v10.11 and later.
(Closes #574, #587)
closed 2020-04-07 by Nightwalker-87