stlink-org/stlink #577: Add "Connect under Reset" option

Gamadril, 2017-03-21:
- [X] Programmer/board type: e.g Stlink/v2-onboard
- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [X] Operating system: Mac OS X
- [X] Stlink tools version and/or git commit hash: 1.3.1-9-gc04df7f
- [X] Stlink commandline tool name: `st-flash`
- [X] Target chip (and optional board): e.g STM32F030F4

SWCLK and SWDIO pins of my mcu are used and configured as output. That means to program the flash I have to use the "Connect Under Reset" option. This option seems to be missing. At least adding --reset does not help in that case.

xor-gate, 2017-03-21:
Connect under reset is not to difficult the stlink has a command to assert the nRST pin and then try to connect. But this needs to be implemented.

iglooom, 2017-04-08:
I have a same trouble and use this workaround [https://github.com/iglooom/stlink/commit/19381ff2e16a121a98184f58dc88838bcb25efd1](url)
It is not fully "connect under reset", but working for me.

xor-gate, 2017-04-08:
The nRST reset pin is assert differently. I'm not sure what the output state of the pin is in your commit. First we need to measure and then implement a fix which works for everybody.

geoffreymbrown, 2020-05-22:
I also needed this functionality.   While I added it to my own copy of the repo, it'd be nice if it made it in somehow.   Here are the issues in stlink_open_usb:

1)  Reset has to be asserted before any mode change can happen
2)  It's somewhat less than optimal that stlink_load_device_params(sl); happens in this routine, since loading flash parameters doesn't seem to work under reset.

Here's what I changed --
1)  reset parameter from bool to number :   values 0,1 keep current behavior.   reset == 2 is used to change the following code : 

` if (stlink_current_mode(sl) != STLINK_DEV_DEBUG_MODE) stlink_enter_swd_mode(sl);

    if (reset) {
        if ( sl->version.stlink_v > 1) stlink_jtag_reset(sl, 2);
        stlink_reset(sl);
        usleep(10000);
    }`

To:
 
`
 if (reset == 2)  stlink_jtag_reset(sl,0);
 if (stlink_current_mode(sl) != STLINK_DEV_DEBUG_MODE) stlink_enter_swd_mode(sl);

    if (reset == 1) {
        if ( sl->version.stlink_v > 1) stlink_jtag_reset(sl, 2);
        stlink_reset(sl);
        usleep(10000);
    }`

Nightwalker-87, 2020-05-22:
@geoffreymbrown: What about proposing a PR?

geoffreymbrown, 2020-05-22:
The various existing apps wouldn't automatically make use of it, but anything using the library could.
I'm happy to propose a narrow PR (just the changes to make this work)
Geoffrey

closed 2020-05-28 by Nightwalker-87