stlink-org/stlink #579: STM32VL won't flash

bolorkhuu, 2017-03-25:
Hello,

I'm getting following error on stm32vl discovery board.

```
> st-flash write STM32VL-Discovery.bin 0x8000000
st-flash 1.3.1
2017-03-25T04:54:29 INFO src/common.c: Loading device parameters....
2017-03-25T04:54:29 INFO src/common.c: Device connected is: F1 Medium/Low-density Value Line device, id 0x10016420
2017-03-25T04:54:29 INFO src/common.c: SRAM size: 0x2000 bytes (8 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 1024 bytes
2017-03-25T04:54:29 INFO src/common.c: Attempting to write 19612 (0x4c9c) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08004c00 erased
2017-03-25T04:54:29 INFO src/common.c: Finished erasing 20 pages of 1024 (0x400) bytes
2017-03-25T04:54:29 INFO src/common.c: Starting Flash write for VL/F0/F3 core id
2017-03-25T04:54:29 INFO src/flash_loader.c: Successfully loaded flash loader in sram
  0/19 pages written2017-03-25T04:54:29 ERROR src/flash_loader.c: write error, count == 511
2017-03-25T04:54:29 ERROR src/common.c: stlink_flash_loader_run(0x8000400) failed! == -1
stlink_fwrite_flash() == -1
```
- [ ] Operating system: Mac OS X Yosemite
- [ ] Stlink tools version: v1.3.1
- [ ] Target chip (and optional board): (STM32VL Discovery)

```
> st-util -p 4242
st-util 1.3.1
2017-03-25T04:58:24 INFO src/common.c: Loading device parameters....
2017-03-25T04:58:24 INFO src/common.c: Device connected is: F1 Medium/Low-density Value Line device, id 0x10016420
2017-03-25T04:58:24 INFO src/common.c: SRAM size: 0x2000 bytes (8 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 1024 bytes
2017-03-25T04:58:24 INFO src/gdbserver/gdb-server.c: Chip ID is 00000420, Core ID is  1ba01477.
2017-03-25T04:58:24 INFO src/gdbserver/gdb-server.c: Listening at *:4242...
```

```
> arm-none-eabi-gdb STM32VL-Discovery.elf 
GNU gdb (GNU Tools for ARM Embedded Processors) 7.10.1.20160923-cvs
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin10 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from STM32VL-Discovery.elf...done.
(gdb) tar extended-remote :4242
Remote debugging using :4242
Reset_Handler () at ../system/src/cortexm/exception_handlers.c:31
31	{
(gdb) load
Loading section .isr_vector, size 0x418 lma 0x8000000
Loading section .inits, size 0x28 lma 0x8000418
Loading section .text, size 0x466d lma 0x8000440
Loading section .data, size 0x1ec lma 0x8004ab0
Error finishing flash operation
(gdb) 
```

Does anyone know, how to solve this?

Thank you,

xor-gate, 2017-03-29:
Thats weird, i'm not sure what causes the problem as the flash loader is written to RAM and then fails at the end.

Faduf, 2018-04-01:
Have you tried with the Boot0 to Vdd before powering the board. This solved the problem for me.

Nightwalker-87, 2020-02-21:
Please verify if the problem still exists in Release v1.6.0 and/or if the feedback by @Faduf resolves the issue.

nerdmaennchen, 2020-03-14:
Just got fresh hardware to play with (AZDelivery STM32 STM32F103C8T6 and a Youmile ST-LINK V2 Emulator) and see exactly this behavior when flashing via st-util as gdb server and arm-none-eabi-gdb as client.

I can flash the target with st-flash without any problem. It seems to be an issue specific to st-util.

How can I help?

nerdmaennchen, 2020-03-14:
i got some more output from st-util with -v99:

~~~
2020-03-14T12:57:10 INFO common.c: Finished erasing 1 pages of 1024 (0x400) bytes
2020-03-14T12:57:10 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_mem32 40 bytes to 0x20000000
2020-03-14T12:57:10 INFO flash_loader.c: Successfully loaded flash loader in sram
2020-03-14T12:57:10 DEBUG common.c: *** stlink_read_debug32 82 is 0x40022010
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_debug32 45670123 to 0x40022004
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_debug32 cdef89ab to 0x40022004
2020-03-14T12:57:10 DEBUG common.c: *** stlink_read_debug32 2 is 0x40022010
2020-03-14T12:57:10 DEBUG common.c: Successfully unlocked flash
2020-03-14T12:57:10 DEBUG common.c: *** stlink_read_debug32 2 is 0x40022010
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_debug32 1 to 0x40022010
2020-03-14T12:57:10 DEBUG common.c: Finished unlocking flash, running loader!
2020-03-14T12:57:10 DEBUG flash_loader.c: Running flash loader, write address:0x8000000, size: 1024
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_mem32 1024 bytes to 0x20000028
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_reg
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_reg
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_reg
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_reg
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_reg
2020-03-14T12:57:10 DEBUG common.c: *** stlink_run ***
2020-03-14T12:57:10 DEBUG common.c: *** stlink_status ***
2020-03-14T12:57:10 DEBUG common.c:   core status: running
2020-03-14T12:57:10 DEBUG common.c: *** stlink_status ***
2020-03-14T12:57:10 DEBUG common.c:   core status: running
2020-03-14T12:57:10 DEBUG common.c: *** stlink_status ***
2020-03-14T12:57:10 DEBUG common.c:   core status: running
[...]
2020-03-14T12:57:10 DEBUG common.c: *** stlink_status ***
2020-03-14T12:57:10 DEBUG common.c:   core status: halted
2020-03-14T12:57:10 DEBUG common.c: *** stlink_read_reg
2020-03-14T12:57:10 DEBUG common.c:  (2) ***
2020-03-14T12:57:10 DEBUG usb.c: r_idx ( 2) = 0x00000000
2020-03-14T12:57:10 DEBUG common.c: *** stlink_read_debug32 1 is 0x40022010
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_debug32 81 to 0x40022010

2020-03-14T12:57:10 INFO common.c: Starting verification of write complete
2020-03-14T12:57:10 DEBUG common.c: *** stlink_read_mem32 ***
2020-03-14T12:57:10 INFO common.c: Flash written and verified! jolly good!
2020-03-14T12:57:10 DEBUG gdb-server.c: flash_do: page 08000400
2020-03-14T12:57:10 INFO common.c: Attempting to write 1024 (0x400) bytes to stm32 address: 134218752 (0x8000400)
2020-03-14T12:57:10 DEBUG common.c: *** stlink_core_id ***
2020-03-14T12:57:10 DEBUG common.c: core_id = 0x1ba01477
2020-03-14T12:57:10 DEBUG common.c: *** stlink_read_debug32 20 is 0x4002200c
2020-03-14T12:57:10 DEBUG common.c: *** stlink_read_debug32 81 is 0x40022010
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_debug32 45670123 to 0x40022004
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_debug32 cdef89ab to 0x40022004
2020-03-14T12:57:10 DEBUG common.c: *** stlink_read_debug32 1 is 0x40022010
2020-03-14T12:57:10 DEBUG common.c: Successfully unlocked flash
2020-03-14T12:57:10 DEBUG common.c: *** stlink_read_debug32 1 is 0x40022010
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_debug32 3 to 0x40022010
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_debug32 8000400 to 0x40022014
2020-03-14T12:57:10 DEBUG common.c: *** stlink_read_debug32 1 is 0x40022010
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_debug32 41 to 0x40022010
2020-03-14T12:57:10 DEBUG common.c: *** stlink_read_debug32 20 is 0x4002200c
2020-03-14T12:57:10 DEBUG common.c: *** stlink_read_debug32 1 is 0x40022010
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_debug32 81 to 0x40022010
Flash page at addr: 0x08000400 erased
2020-03-14T12:57:10 INFO common.c: Finished erasing 1 pages of 1024 (0x400) bytes
2020-03-14T12:57:10 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_mem32 40 bytes to 0x20000000
2020-03-14T12:57:10 INFO flash_loader.c: Successfully loaded flash loader in sram
2020-03-14T12:57:10 DEBUG common.c: *** stlink_read_debug32 81 is 0x40022010
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_debug32 45670123 to 0x40022004
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_debug32 cdef89ab to 0x40022004
2020-03-14T12:57:10 DEBUG common.c: *** stlink_read_debug32 1 is 0x40022010
2020-03-14T12:57:10 DEBUG common.c: Successfully unlocked flash
2020-03-14T12:57:10 DEBUG common.c: *** stlink_read_debug32 1 is 0x40022010
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_debug32 1 to 0x40022010
2020-03-14T12:57:10 DEBUG common.c: Finished unlocking flash, running loader!
2020-03-14T12:57:10 DEBUG flash_loader.c: Running flash loader, write address:0x8000400, size: 1024
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_mem32 1024 bytes to 0x20000028
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_reg
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_reg
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_reg
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_reg
2020-03-14T12:57:10 DEBUG common.c: *** stlink_write_reg
2020-03-14T12:57:10 DEBUG common.c: *** stlink_run ***
2020-03-14T12:57:10 DEBUG common.c: *** stlink_status ***
2020-03-14T12:57:10 DEBUG common.c:   core status: halted
2020-03-14T12:57:10 DEBUG common.c: *** stlink_read_reg
2020-03-14T12:57:10 DEBUG common.c:  (2) ***
2020-03-14T12:57:10 DEBUG usb.c: r_idx ( 2) = 0x000001ff
2020-03-14T12:57:10 ERROR flash_loader.c: write error, count == 511
2020-03-14T12:57:10 ERROR common.c: stlink_flash_loader_run(0x8000400) failed! == -1
2020-03-14T12:57:10 DEBUG gdb-server.c: send: E00
~~~

The problem might be that the core is not runningt when flashing the second sector; I'm just guessing here

nerdmaennchen, 2020-03-14:
before you ask: I'm currently using the master's HEAD (f5d0454ab0a18f209ad5b7d0d51f3945b27dd892)

Nightwalker-87, 2020-03-14:
@nerdmaennchen: Thanks for the feedback. What is the core ID of your STM32F103C8T6? Can you verify that it is not a CKS32F103C8T6 with a fake marking? We currently face several problems with this sort of chinese clone boards (see Release v1.6.1 Tickets) that appeared on the market from Dec 2019. I would like to rule out that we accidentally mix up different issues. Therefore it would be helpful to either be sure of original ST-Microelectronics chips or have another type of board used to analyse this problem, which is not affected by this product counterfeiting.

nerdmaennchen, 2020-03-14:
the reported coreid of the target is in the above log:
~~
2020-03-14T12:57:10 DEBUG common.c: core_id = 0x1ba01477
~~~
frankly, I don't know if it's genuine or not (it was very cheap though)

Nightwalker-87, 2020-03-14:
Looks as if you are lucky, though you caught a cheap board. The affected boards mentioned have Core ID 0x2ba01477, so we should be fine then. Thanks for verifying. :+1: We will keep up with this then, please give us some time to solve this.

nerdmaennchen, 2020-03-14:
Of course, take the time you need!
Also I'd like to express my appreciation for your project: 

Kudos! I've been using your tools since five years or so and they are sill my go-to-tools
You rock!

If you need any help, I'd be delighted to do my part.

Nightwalker-87, 2020-03-14:
It appears that we need a closer look at what happens step by step when `flash_loader.c` is called to further encircle the issue. @slyshykO: Are you interested to look at this on your Windows system? As I only have a non-genuine CKS32F103C8T6 chip I may only look into the code, but without the ability to test and debug properly. :-/

nerdmaennchen, 2020-03-15:
I did some debugging myself; the problem seems to be that the first word of the second page cannot be written and the flash loader (in sram) bails out due to the PGERR-bit beeing set.

This happens when a writing into the flash without previously erasing the corresponding page.
And as it happens there is no code in common.c (around line 2377) that would erase the corresponding page.
When I do a "clean" mass erase on the entire flash memory and then load my binary via the above mentioned method, everything works.
However, this only works when flipping bits from 1 to 0 in flash memory (e.g., after a fresh erase or when programming a bit pattern that is already stored).

I do have the strong feeling that I somehow maneuvred myself in a situation where I have "magically" programmed the first page of my target and stopped there; when I later tried to reprogram the device via st-util I've seen the above mentioned behavior.
All I'm saying is: The way I got to the situation is a bit odd.

Anyways: The fix seems rather easy: erase the page that wants to be programmed before programing the page. This should be done just above line 2337 in common.c.
Since the code there seems to handle stm32f0 cores as well I'd expect users of those ICs to see the same issues. Can anybody confirm?

nerdmaennchen, 2020-03-15:
For some reason the PER bit is set when writing the first word to the second page.
The code that performs page erases before programming seems is executed and is written exactly as stated in the Programming Manual (PM0075).

However, if I do a mass erase (with st-flash) before programming the device via st-util and gdb everything works smoothly.

nerdmaennchen, 2020-03-15:
found the problem.
the PG bit might be set before doing the page erase.
when the START bit is set later the PG as well as the PER bits will be set and the flash controller will remain in the PG state and not do a page erase at all.
That explains why the first page was programmable but the second wasn't, since the flash loader set the PG bit during programming of the first page.

cheers

Nightwalker-87, 2020-03-15:
@nerdmaennchen: Thank you so much for your contribution. I'll review the PR as soon as possible. You were on linux, right? Can someone else test this with another board (and eventally a different OS) as well - ticket was opened for macOS BTW) to have redundant verification? @bmarvo, @bolorkhuu or @Faduf maybe?

nerdmaennchen, 2020-03-15:
jep, im an Arch-fanboy (Linux) and I don't have easy acess to other OSes.

However, IMHO the fix is entirely OS independent as was the cause for the problem.

Nightwalker-87, 2020-03-15:
That's what I thought as well, but I feel like that it is always good to have verification on at least a second setup. Don't bother about this too much, you are fine and I believe you spent enough valuable efforts on this. Let somebody else take a look at it and relax ;-)

nerdmaennchen, 2020-03-15:
very well then.

relax mode:
ACTIVATED

don't worry about my time. I actually enjoyed it quite a lot :D

Nightwalker-87, 2020-03-18:
@nerdmaennchen: You may have a look at other issues in the meanwhile if you fancy.

PR #876 (2020-03-19): clear the PG bit before setting the PER bit
after the flash loader ran the PG bit is still set in the CR register.
in order to program the next page that page needs to be erased first and to get that working the PG bit has to be cleared so the flash controller knows what to do.
this fixes #579
commit 268313bc (2020-03-19): clear the PG bit before setting the PER bit

after the flash loader ran the PG bit is still set in the CR register.
in order to program the next page that page needs to be erased first and to get that working the PG bit has to be cleared so the flash controller knows what to do.
this fixes #579
closed 2020-03-19 by Nightwalker-87