stlink-org/stlink #585: improved error message request for usb.c

petertorelli, 2017-04-10:
Ubuntu 16.10 requires root access to open a device. I often forget this since I routinely switch between macOS, Windows and Linux. This is a bit pedantic, but the WLOG messages in src/usb.c aren't succinct enough or are incorrect. I changed them to the following as a user-experience improvement, but I don't have branch permissions:

````
diff --git a/src/usb.c b/src/usb.c
index 03ac1ab..33bb350 100644
--- a/src/usb.c
+++ b/src/usb.c
@@ -959,7 +959,7 @@ static size_t stlink_probe_usb_devs(libusb_device **devs, stlink_t **sldevs[]) {
         struct libusb_device_descriptor desc;
         int r = libusb_get_device_descriptor(dev, &desc);
         if (r < 0) {
-            WLOG("failed to get libusb device descriptor\n");
+            WLOG("failed to get libusb device descriptor during count (libusb error: %d)\n", ret);
             break;
         }
 
@@ -983,7 +983,7 @@ static size_t stlink_probe_usb_devs(libusb_device **devs, stlink_t **sldevs[]) {
         struct libusb_device_descriptor desc;
         ret = libusb_get_device_descriptor(dev, &desc);
         if (ret < 0) {
-            WLOG("failed to get libusb device descriptor\n");
+            WLOG("failed to get libusb device descriptor during open (libusb error: %d)\n", ret);
             break;
         }
 
@@ -997,7 +997,10 @@ static size_t stlink_probe_usb_devs(libusb_device **devs, stlink_t **sldevs[]) {
 
         ret = libusb_open(dev, &handle);
         if (ret < 0) {
-            WLOG("failed to get libusb device descriptor\n");
+            if (ret == LIBUSB_ERROR_ACCESS)
+                WLOG("failed to open USB device (LIBUSB_ERROR_ACCESS), try running as root?\n");
+            else
+                WLOG("failed to open USB device (libusb error: %d)\n", ret);
             break;
         }
````

xor-gate, 2017-04-11:
Please use a Pull Request to submit contributions. Thanks!

petertorelli, 2017-04-12:
@xor-gate I'd love to but I don't have permissions to make a branch to make a pull request. :)

xor-gate, 2017-04-12:
Hi @petertorelli, 

Probably you are not used to Git workflow -> Fork repo (github) -> Create feature Branch -> Create Pull request to upstream -> Merge.

You should read some documentation, because I only accept PRs.

https://help.github.com/articles/creating-a-pull-request-from-a-fork/

Thank you,
Jerry

xor-gate, 2017-05-09:
ping @petertorelli

xor-gate, 2017-05-15:
Closing due too inactivity, feel free to comment and provide a pull request.

Nightwalker-87, 2020-03-19:
@petertorelli: Please submit this proposal in form of a PR so that it can be properly addressed.

PR #883 (2020-03-21): Added more error info to WLOGs during probe.
Closes #585.
closed 2020-03-21 by Nightwalker-87