stlink-org/stlink #586: STM32L151UC - stlink_read_mem32() does not seem to return correct values

DanielAW, 2017-04-11:
- [X] Programmer/board type: Stlink/v2
- [X] Programmer firmware version: V2.J28.S7 STM32+STM8 Debugger
- [X] Operating system: Arch-Linux
- [X] Stlink tools version: 1.3.1
- [X] Stlink commandline tool name: `st-util`
- [X] Target chip: e.g STM32L151UC

I try to debug a STM32 chip without any input file for `gdb`. Connecting to the chip works fine, but as soon as I try to read memory I get strange results. This also means that the the disassembled instructions make no sense. 
I've got a "clean" dump from the official ST-Link Untility which I can use for comparison. The executed instructions (`stepi`) seems to make sense, also the register values are ok. 
Only the memory does not make sense to me. Is it possible that the memory is read from a wrong offset?

The following is output at the very start of debugging (after a hard reset)
Output `st-util -v99`:

```
2017-04-11T10:49:22 DEBUG src/gdbserver/gdb-server.c: send: l
2017-04-11T10:49:22 DEBUG src/gdbserver/gdb-server.c: recv: m8000082,4
2017-04-11T10:49:22 DEBUG src/common.c: *** stlink_read_mem32 ***
2017-04-11T10:49:22 DEBUG src/gdbserver/gdb-server.c: send: 00000000
2017-04-11T10:49:22 DEBUG src/gdbserver/gdb-server.c: recv: p19
2017-04-11T10:49:22 DEBUG src/common.c: *** stlink_read_reg
2017-04-11T10:49:22 DEBUG src/common.c:  (16) ***
2017-04-11T10:49:22 DEBUG src/usb.c: r_idx (16) = 0x01000000
2017-04-11T10:49:22 DEBUG src/gdbserver/gdb-server.c: send: 00000001
2017-04-11T10:49:22 DEBUG src/gdbserver/gdb-server.c: recv: m0,4
2017-04-11T10:49:22 DEBUG src/common.c: *** stlink_read_mem32 ***
2017-04-11T10:49:22 DEBUG src/gdbserver/gdb-server.c: send: 00000000
2017-04-11T10:49:30 DEBUG src/gdbserver/gdb-server.c: recv: m8000082,4
2017-04-11T10:49:30 DEBUG src/common.c: *** stlink_read_mem32 ***
2017-04-11T10:49:30 DEBUG src/gdbserver/gdb-server.c: send: 00000000
```
Output of `gdb`

```
Remote debugging using 192.168.56.1:4242
0x08000082 in ?? ()
(gdb) x 0x08000082
0x8000082:	0x00000000
```
The instruction on this memory location is not 0x0

xor-gate, 2017-04-11:
I think the core is not halted when reading those regs.

DanielAW, 2017-04-11:
Hmm ok, but as I sayed: Single instruction stepping works fine. The executed instructions generate the expected output.

Nightwalker-87, 2020-03-21:
@DanielAW: What is the current state of this? Have you found a solution?

DanielAW, 2020-03-26:
I think I just worked with `stepi` sorry I submitted this issue a long time ago.

Nightwalker-87, 2020-03-27:
Closing this ticket as the given feedback seems to explain the observed behaviour.

closed 2020-03-27 by Nightwalker-87