stlink-org/stlink #587: Can't connect to ST Link V1 / STM32VL Discovery on Mac OSX Sierra

C44Supra, 2017-04-18:
I've been trying to get st-link going with my STM32VL Discovery board under Mac OSX Sierra (10.12.1). I have built st-link both from source and installed it with homebrew (but obviously removed one before I continued with the other). Both times I've encountered the same problem:
```
$ st-util -1
st-util 1.3.1
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
2017-04-18T15:21:43 WARN src/sg.c: libusb_claim_interface() failed
2017-04-18T15:21:43 ERROR src/sg.c: Could not open stlink device
```

I've have installed the kext / driver from the "stlinkv1_macosx_driver" folder (for which I had to modify the last 10.10 case in install.sh to get it to install at all). I have the kext-dev-mode flag set although I have no idea how viable of a solution that still is under Sierra.

For good measure I rebooted, reconnected the board and tried again and got a different result:
```
$ st-util -1
st-util 1.3.1
2017-04-18T15:04:32 WARN src/sg.c: Failed to find an stlink v1 by VID:PID
2017-04-18T15:04:32 ERROR src/sg.c: Could not open stlink device
```

I checked the system profiler to see if the ST Link device showed up there, and it does:
[image]

I've googled the VendorID and ProductID and those seem to be correct for an ST Link V1.

After that I installed Ubuntu 17.04 in a VM under VMWare Fusion and built st-link from source, reconnected the board to the system, was asked by VMWare if I wanted to connect it to the VM and chose yes. Ran st-util -1 and the whole thing worked flawlessly.

Is there something I might be doing wrong on the OSX side of things? Is there perhaps a different kext instead of the 10.10 version?

To recap: 

- stlink built from a fresh git clone on OSX
- can't claim usb device without kext installed
- can't find device with kext installed
- works perfectly fine using the same git clone under Ubuntu in a VM on the same exact system (no usb shenanigans necessary)

xor-gate, 2017-04-18:
Duplicate of #574, feel free to comment or proceed in the other issue.

Nightwalker-87, 2020-04-07:
Closed by commit 59c5cfd7ec5c7f2a035944ecc5ca01d3d7b31170.

closed 2020-04-07 by Nightwalker-87