stlink-org/stlink #589: st-flash failed writing flash, but could succeed with --debug.

assiss, 2017-05-07:
- [X] Programmer/board type:  Stlink/v2
- [X] Operating system: Arch Linux
- [X] Stlink tools version and/or git commit hash: stlink 1.3.1
- [X] Stlink commandline tool name: st-flash
- [X] Target chip (and optional board): STM32F030F4p6. SSOP20, single chip with SSOP20_to_DIP breadbord.

st-flash write main.bin 0x8000000 

Output:

```
st-flash 1.3.1
2017-05-07T19:47:42 INFO src/common.c: Loading device parameters....
2017-05-07T19:47:42 INFO src/common.c: Device connected is: F0 small device, id 0x10006444
2017-05-07T19:47:42 INFO src/common.c: SRAM size: 0x1000 bytes (4 KiB), Flash: 0x4000 bytes (16 KiB) in pages of 1024 bytes
2017-05-07T19:47:42 INFO src/common.c: Attempting to write 4772 (0x12a4) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08001000 erased
2017-05-07T19:47:42 INFO src/common.c: Finished erasing 5 pages of 1024 (0x400) bytes
2017-05-07T19:47:42 INFO src/common.c: Starting Flash write for VL/F0/F3 core id
2017-05-07T19:47:42 INFO src/flash_loader.c: Successfully loaded flash loader in sram
  0/4 pages written2017-05-07T19:47:46 ERROR src/flash_loader.c: flash loader run error
2017-05-07T19:47:46 ERROR src/common.c: stlink_flash_loader_run(0x8000400) failed! == -1
stlink_fwrite_flash() == -1
```

run 10 times may succeeded once.

but when use --debug:
st-flash --debug write main.bin 0x8000000

It almost succeeded every time(99%).

st-info and st-utils worked well.

Bye the way, openocd would connected and supported debug, but could not flash either.
OpenOCD complained:
```
Error erasing flash with vFlashErase packet
```

Now only stm32loader.py with uart succeeded 100%.

assiss, 2017-05-07:
After adding a 47uF capacitor between VDD and VSS, st-flash works very well.
Don't know why. Without the capacitor, STM32F030F4P6 still runs very well.

assiss, 2017-05-08:
It is my falt. I am new to stlink, and made some stupid mistakes.

Supplying the mcu with independent power source, and no problems at all.

please close this.

xor-gate, 2017-05-08:
You could close yourself. But thanks for reporting. Good luck!

closed 2017-05-08 by xor-gate