stlink-org/stlink #590: Warnings/error break build for gcc 7

atu-guda, 2017-05-07:
- [x ] Programmer/board type: Stlink/v2, Stlink/v2-onboard, Stlink/V2-1 onboard
- [X ] Operating system: Linux Fedora 26+
- [X ] Stlink tools version v1.3.1/git-cc3a5705099019e01dc59f7907d0c5b1b85878d8

During compilation with gcc 7.0.1, some warnings about fall through (missing break)
is reported. This warnings is treated as errors, so make fails to build.

Output:
```
stlink-20170507/src/gdbserver/gdb-server.c:1626:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
                     case '4': { // insert access watchpoint

stlink-20170507/src/gdbserver/gdb-server.c:1663:27: error: this statement may fall through [-Werror=implicit-fallthrough=]
                         if(delete_data_watchpoint(sl, addr) < 0) {
```

A patch is ready. Will try to submit it.

atu-guda, 2017-05-07:
[stlink-addbreak.patch.txt](https://github.com/texane/stlink/files/981737/stlink-addbreak.patch.txt)

Sorry for '.txt' suffix, github do not support '.patch' files from here.

xor-gate, 2017-05-08:
Please submit a git pull request, we don't accept verbatim patches.

commit 5c10d4b8 (2017-05-08): Fix compilation warning for GCC 7, fixes #590
closed 2020-03-18 by Nightwalker-87