stlink-org/stlink #593: Nucleo-L073RZ won't flash

GordonHJSmith, 2017-05-15:
Hi, 

I ran into troubles to flash my nucleo-l073rz with st-flash and st-util/gdb:

- [X] Programmer/board type: Stlink/v2-onboard
- [X] Programmer firmware version: 2.28.18
- [X] Operating system: Mac OS X (10.12.4)
- [X] Stlink tools version and/or git commit hash: 5c10d4b83a4703edb492c76f1427c4a4e1a568b1
- [X] Stlink commandline tool name:`st-flash`, `st-util`
- [X] Target chip (and optional board): STM32L073RZ on Nucleo

st-flash output:

```
./st-flash --debug write fw.bin 0x8000000
...
169/170 pages written2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_write_debug32 2000084c to 0x80054fc
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_read_debug32 10c is 0x40022018
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_write_debug32 2000084c to 0x8005500
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_read_debug32 10c is 0x40022018
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_write_debug32 20000854 to 0x8005504
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_read_debug32 10c is 0x40022018
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_write_debug32 20000854 to 0x8005508
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_read_debug32 10c is 0x40022018
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_write_debug32 20000 to 0x800550c
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_read_debug32 10c is 0x40022018
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_write_debug32 ffffffff to 0x8005510
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_read_debug32 10c is 0x40022018

2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_read_debug32 7 is 0x40022004
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_write_debug32 7 to 0x40022004
2017-05-15T14:57:16 INFO src/common.c: Starting verification of write complete
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_read_mem32 ***
data_len = 128 0x80
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

2017-05-15T14:57:16 ERROR src/common.c: Verification of flash failed at offset: 0
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x8000000
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_write_reg
data_len = 2 0x2
 80 00

2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x8000004
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_write_reg
data_len = 2 0x2
 80 00

2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_run ***
stlink_fwrite_flash() == -1
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_exit_debug_mode ***
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_write_debug32 a05f0000 to 0xe000edf0
2017-05-15T14:57:16 DEBUG src/common.c: *** stlink_close ***
```

Same issue withh st-util and gdb.

gdb output:
```
(gdb) load
Loading section .text, size 0x4888 lma 0x8000000
Loading section .init_array, size 0x10 lma 0x8004888
Loading section .ARM.extab, size 0x1d0 lma 0x8004898
Loading section .ARM.exidx, size 0x240 lma 0x8004a68
Loading section .data, size 0x86c lma 0x8004ca8
Error finishing flash operation
```
st-util output:
```
st-util 1.3.1-17-g5c10d4b
2017-05-15T14:59:05 INFO src/common.c: Loading device parameters....
2017-05-15T14:59:05 INFO src/common.c: Device connected is: L0x Category 5 device, id 0x20006447
2017-05-15T14:59:05 INFO src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x30000 bytes (192 KiB) in pages of 128 bytes
2017-05-15T14:59:05 INFO src/gdbserver/gdb-server.c: Chip ID is 00000447, Core ID is  0bc11477.
2017-05-15T14:59:05 INFO src/gdbserver/gdb-server.c: Listening at *:4242...
2017-05-15T14:59:11 INFO src/gdbserver/gdb-server.c: Found 4 hw breakpoint registers
2017-05-15T14:59:11 INFO src/gdbserver/gdb-server.c: GDB connected.
2017-05-15T14:59:17 INFO src/common.c: Attempting to write 128 (0x80) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000000 erased
2017-05-15T14:59:17 INFO src/common.c: Finished erasing 1 pages of 128 (0x80) bytes
2017-05-15T14:59:17 INFO src/common.c: Starting Half page flash write for STM32L core id
2017-05-15T14:59:17 INFO src/flash_loader.c: Successfully loaded flash loader in sram
2017-05-15T14:59:18 ERROR src/flash_loader.c: write error, count == 4294967295
2017-05-15T14:59:18 WARN src/common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
2017-05-15T14:59:18 WARN src/common.c:
write_half_pages failed == -1
  0/  1 pages written
2017-05-15T14:59:18 INFO src/common.c: Starting verification of write complete
2017-05-15T14:59:18 ERROR src/common.c: Verification of flash failed at offset: 0
```
In both configurations I tried to do an erase beforehand with st-flash (which is working), but the output is the same.

This issue seems to be related to #579.

Thank you for your help, I'm available to make further tests.

xor-gate, 2017-05-15:
Strange, it looks this is not good:

```
2017-05-15T14:59:18 ERROR src/flash_loader.c: write error, count == 4294967295
2017-05-15T14:59:18 WARN src/common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
```

An incorrect count is reported, and it is using the L1 flash loader for some reason.

Faduf, 2018-04-01:
Have you tried with the Boot0 to Vdd before powering the board. This solved the problem for me.

GordonHJSmith, 2018-04-01:
I don't have the board anymore.
Anyway it was working on another same board reference.

The original board wasn't new when I got it, maybe the option bytes were configured with a bad watchdog or something.

spth, 2018-04-03:
I see the same problem on two STM32L073NUCLEO-L073RZ boards. Both boards are new. Writing the flash via st-link worked at least once for each board, but soon after I only kept getting errors:

~~~~
root@notebook5:/home/philipp/stlink/build/Release# ./st-flash --reset --format ihex write /home/philipp/stdcbench-trunk/stdcbench/stdcbench.ihx 
st-flash 1.4.0-19-gc0d759d
2018-04-03T11:05:36 INFO common.c: Loading device parameters....
2018-04-03T11:05:36 INFO common.c: Device connected is: L0x Category 5 device, id 0x20086447
2018-04-03T11:05:36 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x30000 bytes (192 KiB) in pages of 128 bytes
2018-04-03T11:05:36 INFO common.c: Attempting to write 532 (0x214) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000200 erased
2018-04-03T11:05:36 INFO common.c: Finished erasing 5 pages of 128 (0x80) bytes
2018-04-03T11:05:36 INFO common.c: Starting Half page flash write for STM32L core id
2018-04-03T11:05:36 INFO flash_loader.c: Successfully loaded flash loader in sram
2018-04-03T11:05:39 ERROR flash_loader.c: flash loader run error
2018-04-03T11:05:39 WARN common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
2018-04-03T11:05:39 WARN common.c: 
write_half_pages failed == -1
  3/  4 pages written
2018-04-03T11:05:40 INFO common.c: Starting verification of write complete
2018-04-03T11:05:40 ERROR common.c: Verification of flash failed at offset: 0
stlink_fwrite_flash() == -1
~~~~

Output with --debug is the same as in the original bug report.

Philipp

Nightwalker-87, 2020-03-20:
@spth @Faduf : What is the current state of this? Have you found anything out apart from the workaround?

Nightwalker-87, 2020-04-07:
Please check if #876 resolves this problem as it resolves the related issue #579 mentioned in the opening post.

PR #1113 (2021-03-25): Flash loader rework
In this PR, the waiting for data writing is changed from cycles to ms. This should be a more robust solution. 
Fixed a flash loader bug. The flash loader had an incorrect busy wait cycle. The use of bootloader argument registers during boot removed. Removed magic with recalculation of the `size` to the `count`. The number of bytes to load (`size`) is transferred to the flash loader now.
Added disabling of interrupts during flash loader execution. In some cases interrupts caused a failure

(Closes #356, Closes #556, Closes #593, Closes #597, Closes #607, Closes #612, Closes  [... cut]
closed 2021-03-25 by Nightwalker-87