stlink-org/stlink #596: Cannot flash STM32L151 w Stlink/v2 on MAC

JonasWestAlro, 2017-05-19:
Hi, 

I am trying to flash a an STM32L151 using Stlink/v2 on a Mac with OSX Sierra.
I've installed stlink 1.3.1 using homebrew as explained in the readme. I am now trying to flash the device: 

```
$ st-flash write Bootloader.bin 0x08000000
st-flash 1.3.1
2017-05-19T14:13:37 INFO src/common.c: Loading device parameters....
2017-05-19T14:13:37 INFO src/common.c: Device connected is: L1 Medium-Plus-density device, id 0x10386427
2017-05-19T14:13:37 INFO src/common.c: SRAM size: 0x8000 bytes (32 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 256 bytes
2017-05-19T14:13:37 INFO src/common.c: Attempting to write 37348 (0x91e4) bytes to stm32 address: 134217728 (0x8000000)
2017-05-19T14:13:37 WARN src/common.c: pecr.pelock not clear (0x7)
2017-05-19T14:13:37 ERROR src/common.c: Failed to erase_flash_page(0x8000000) == -1
stlink_fwrite_flash() == -1
```

Anyone?

xor-gate, 2017-06-10:
Could you try a building a newer version from the master branch?

Nightwalker-87, 2020-03-21:
@JonasWestAlro: What is the current state of this? Have you found a solution in the meanwhile? Can you rule out any hardware related issues? Please let us know.

Nightwalker-87, 2020-04-09:
> 2017-05-19T14:13:37 WARN src/common.c: pecr.pelock not clear (0x7)

This indicates that the flash memory of the used device is locked.
Unlocking for this chip is possible since commit 302dc872f4b73f13a94ff2e9f0359c4195f6279c, which added support to write option bytes for the STM32L1 device family. An example how to make use of this feature with the `st-flash` command can be found in our project documentation in `/doc/tutorial.md`.
As this answer fully addresses this request for support, this issue is now closed as resolved.

closed 2020-04-09 by Nightwalker-87