stlink-org/stlink #597: There is rare case that st-util can't flash firmware onto NUCLEO-F091RC

master-q, 2017-05-20:
There is rare case that st-util can't flash firmware onto NUCLEO-F091RC.

- [X] Programmer/board type: Stlink/v2-onboard
- [X] Programmer firmware version: v1.3.1-19-g55c0572
- [X] Operating system: Debian GNU/Linux sid
- [X] Stlink tools version and/or git commit hash: 55c057296ad15fa6de9909206098bd4eed8f6311
- [X] Stlink commandline tool name: `st-util`
- [X] Target chip (and optional board): STM32F091RCT6 on NUCLEO-F091RC

`st-util` output:

```
$ sudo st-util
--snip--
2017-05-20T17:04:06 ERROR src/gdbserver/gdb-server.c: cannot recv: -2
2017-05-20T17:04:07 INFO src/gdbserver/gdb-server.c: Listening at *:4242...
2017-05-20T17:04:07 INFO src/gdbserver/gdb-server.c: Found 4 hw breakpoint registers
2017-05-20T17:04:07 INFO src/gdbserver/gdb-server.c: GDB connected.
2017-05-20T17:04:08 INFO src/gdbserver/gdb-server.c: Found 4 hw breakpoint registers
2017-05-20T17:04:09 INFO src/common.c: Attempting to write 2048 (0x800) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000000 erased
2017-05-20T17:04:09 INFO src/common.c: Finished erasing 1 pages of 2048 (0x800) bytes
2017-05-20T17:04:09 INFO src/common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2017-05-20T17:04:09 INFO src/flash_loader.c: Successfully loaded flash loader in sram
2017-05-20T17:04:13 ERROR src/flash_loader.c: flash loader run error
2017-05-20T17:04:13 ERROR src/common.c: stlink_flash_loader_run(0x8000000) failed! == -1
```

`gdb` output:

```
$ cat ../gdbwrite.boot
tar ext localhost:4242
monitor reset halt
load
monitor reset halt
$ arm-none-eabi-gdb -x ../gdbwrite.boot build/ch.elf
GNU gdb (7.12-6+9+b2) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/ch.elf...done.
_idle_thread.lto_priv.58 (p=0x0) at ../../../ChibiOS_16.1.6/os/rt/src/chsys.c:65
65      static void _idle_thread(void *p) {
Loading section startup, size 0xc0 lma 0x8000000
Loading section .text, size 0x62fc lma 0x80000c0
Loading section .data, size 0xb8 lma 0x80063c0
../gdbwrite.boot:3: Error in sourced command file:
Error finishing flash operation
```

Expected/description: I think gdb can do flash everytime.

xor-gate, 2017-06-10:
Could you try a little older commit from the master branch? Probably there is something wrong with the flash locking/unlocking.

master-q, 2017-06-10:
O.K. I'll try that. Please give me some times....

Nightwalker-87, 2020-03-20:
@master-q: Has there been any further progress on this (e.g. a test with Release v1.4.0) ?

Nightwalker-87, 2020-07-05:
This should be tested again with the `v1.5.0-patch` branch.

rileylyman, 2020-08-20:
I'm having the exact same issue on v1.6.1

Nightwalker-87, 2021-03-24:
@Ant-ON Can you check this one as well? I think though `st-util` was used, this may also be addressed by #1113.

Ant-ON, 2021-03-25:
@Nightwalker-87 Standard error descriptions are not informative. Most likely this issue is closed by #1113.

Nightwalker-87, 2021-03-25:
Yeah, true. However, as the error code
```
2017-05-20T17:04:13 ERROR src/flash_loader.c: flash loader run error
2017-05-20T17:04:13 ERROR src/common.c: stlink_flash_loader_run(0x8000000) failed! == -1
```
is identical to what we've seen before one should be safe to assume the issue is fixed by now.

Ant-ON, 2021-03-25:
@Nightwalker-87 Yes, otherwise someone with the error can open a new issue with a new log

PR #1113 (2021-03-25): Flash loader rework
In this PR, the waiting for data writing is changed from cycles to ms. This should be a more robust solution. 
Fixed a flash loader bug. The flash loader had an incorrect busy wait cycle. The use of bootloader argument registers during boot removed. Removed magic with recalculation of the `size` to the `count`. The number of bytes to load (`size`) is transferred to the flash loader now.
Added disabling of interrupts during flash loader execution. In some cases interrupts caused a failure

(Closes #356, Closes #556, Closes #593, Closes #597, Closes #607, Closes #612, Closes  [... cut]
closed 2021-03-25 by Nightwalker-87