stlink-org/stlink #598: Nucleo STM32L011 Unknown Chip ID: 0x10086457

jchidley, 2017-05-29:
**NOTICE: The issue may be closed without notice when not enough information is provided!**

- [ X]  Stlink/v2-onboard
- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [ X] Operating system: Windows 10
- [ X] Stlink tools version v1.3.0
- [ X] `st-util`
- [ X] Target chip (and optional board): STM32L011K4T6 (STM32L011 Nucleo-32)

Board firmware: V2.J28.M18 STM32 Debug+Mass storage

Output:

```
C:\tools\stlink-1.3.0-win64\bin>st-util
st-util 1.3.0
2017-05-29T14:24:15 INFO C:\Users\Jerry\Desktop\stlink-master\src\common.c: Loading device parameters....
2017-05-29T14:24:15 WARN C:\Users\Jerry\Desktop\stlink-master\src\common.c: unknown chip id! 0x10086457
```

```
C:\tools\stlink-1.3.0-win64\bin>st-info --chipid
libusb: warning [init_device] could not get node connection information for device '\\.\USB#VID_045E&PID_0745&MI_02#8&12437C26&0&0002': [87] The parameter is incorrect.
```

Expected/description:
I expect to see a connection like this, from my working STM32L031 Nucleo-32 (STM32L031K6T6)

```
C:\tools\stlink-1.3.0-win64\bin>st-util
st-util 1.3.0
2017-05-29T14:29:19 INFO C:\Users\Jerry\Desktop\stlink-master\src\common.c: Loading device parameters....
2017-05-29T14:29:19 INFO C:\Users\Jerry\Desktop\stlink-master\src\common.c: Device connected is: L0x Category 2 device, id 0x20186425
2017-05-29T14:29:19 INFO C:\Users\Jerry\Desktop\stlink-master\src\common.c: SRAM size: 0x2000 bytes (8 KiB), Flash: 0x8000 bytes (32 KiB) in pages of 128 bytes
2017-05-29T14:29:19 INFO C:\Users\Jerry\Desktop\stlink-master\src\gdbserver\gdb-server.c: Chip ID is 00000425, Core ID is  0bc11477.
2017-05-29T14:29:19 INFO C:\Users\Jerry\Desktop\stlink-master\src\gdbserver\gdb-server.c: Listening at *:4242...
```

xor-gate, 2017-05-29:
Already supported in v1.3.1 or higher, but no build for windows is available (yet). See https://github.com/texane/stlink/blob/master/ChangeLog.md#v131.

closed 2017-05-29 by xor-gate

jchidley, 2017-05-29:
This explains why I could see no problem in the current source.