stlink-org/stlink #599: STM32F0DISCOVERY Memory Map problem on using gdb/st-link

jchidley, 2017-05-29:
**NOTICE: The issue may be closed without notice when not enough information is provided!**

Thank you for giving feedback to the stlink project. Take some time to fill out
 check boxes with a X in the following items so developers and other people can try to
 to find out what is going on. And add/remove what is appropriate to your problem.

When submitting a feature request, try to reuse the list and add/remove what is appropriate.
 Place a `X` between the brackets `[X]` to mark the list item.

- [ X] Stlink/v2-onboard
 Programmer firmware version: e.g STSW-LINK007 2.27.15
- [ X] Operating system: Windows 10
- [ X] Stlink tools version v1.3.0
- [ X] st-util
- [ X] Target chip (and optional board): STM32F051R8 (STM32F0DISCOVERY)

Output:

```
C:\tools\stlink-1.3.0-win64\bin>st-util
st-util 1.3.0
2017-05-29T18:45:25 INFO C:\Users\Jerry\Desktop\stlink-master\src\usb.c: -- exit_dfu_mode
2017-05-29T18:45:25 INFO C:\Users\Jerry\Desktop\stlink-master\src\common.c: Loading device parameters....
2017-05-29T18:45:25 INFO C:\Users\Jerry\Desktop\stlink-master\src\common.c: Device connected is: F0 device, id 0x20006440
2017-05-29T18:45:25 INFO C:\Users\Jerry\Desktop\stlink-master\src\common.c: SRAM size: 0x2000 bytes (8 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2017-05-29T18:45:25 INFO C:\Users\Jerry\Desktop\stlink-master\src\gdbserver\gdb-server.c: Chip ID is 00000440, Core ID is  0bb11477.
2017-05-29T18:45:25 INFO C:\Users\Jerry\Desktop\stlink-master\src\gdbserver\gdb-server.c: Listening at *:4242...
```

```
C:\Users\jackc\Documents\blink>arm-none-eabi-gdb build\blink.elf
GNU gdb (GNU Tools for ARM Embedded Processors 6-2017-q1-update) 7.12.1.20170215-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-w64-mingw32 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build\blink.elf...done.
(gdb) set debug remote 1
(gdb) target extended-remote :4242
Remote debugging using :4242
Sending packet: $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+#df...Ack
Packet received: PacketSize=3fff;qXfer:memory-map:read+
Packet qSupported (supported-packets) is supported
Sending packet: $vMustReplyEmpty#3a...Ack
Packet received:
Sending packet: $!#21...Ack
Packet received: OK
Sending packet: $Hg0#df...Ack
Packet received:
Sending packet: $qTStatus#49...Ack
Packet received:
Packet qTStatus (trace-status) is NOT supported
Sending packet: $?#3f...Ack
Packet received: S05
Sending packet: $qfThreadInfo#bb...Ack
Packet received:
Sending packet: $qL1200000000000000000#50...Ack
Packet received:
Sending packet: $Hc-1#09...Ack
Packet received:
Sending packet: $qC#b4...Ack
Packet received:
Sending packet: $qAttached#8f...Ack
Packet received:
Packet qAttached (query-attached) is NOT supported
Sending packet: $qOffsets#4b...Ack
Packet received:
Sending packet: $Hg0#df...Ack
Packet received:
Sending packet: $g#67...Ack
Packet received: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffd8fffffff9fffffffeffffff
Sending packet: $g#67...Ack
Packet received: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffd8fffffff9fffffffeffffff
Sending packet: $qL1200000000000000000#50...Ack
Packet received:
Sending packet: $qXfer:memory-map:read::0,fff#1c...Ack
Packet received: m<?xml version="1.0"?><!DOCTYPE memory-map PUBLIC "+//IDN gnu.org//DTD GDB Memory Map V1.0//EN"     "http://sourceware.org/gdb/gdb-memory-map.dtd"><memory-map>  <memory type="rom" start="0x00000000" length="0xzx"/>  <memory type="ram" start="0x20000000" length="0xzx"/>  <memory type="flash" start="0x08000000" length="0xzx">    <property name="blocksize">0xzx</property>  </memory>  <memory type="ram" start="0x40000000" length="0x1fffffff"/>  <memory type="ram" start="0xe0000000" length="0x1fffffff"/>  <memory type="rom" start="0x00010000" length="0xzx"/>  <memory type="rom" start="0x1ffff800" length="0x10"/></memory-map>
Sending packet: $qXfer:memory-map:read::272,d8d#55...Ack
Packet received: l
warning: while parsing target memory map (at line 1): Can't convert length="0xzx" to an integer
Sending packet: $mfffffffe,4#fc...Ack
Packet received: ffffffff
Sending packet: $p19#da...Ack
Packet received: 030000c1
Packet p (fetch-register) is supported
Sending packet: $mfffffffe,2#fa...Ack
Packet received: ffff
0xfffffffe in ?? ()
Sending packet: $qSymbol::#5b...Ack
Packet received:
Packet qSymbol (symbol-lookup) is NOT supported
(gdb)
```

Expected/description:
The memory map entries need to have a length on octal.  
```
<memory type="rom" start="0x00000000" length="0xzx"/>
```

This also seems to happen with 2 other boards:

- Nucleo L302 (STM32F302r8t6)
- Nucleo L031 (STM32L031K6T6)

Jack

jchidley, 2017-05-29:
Works if st-link is run from my Rasperry Pi 2 (server) 1.3.1.  Still using the gdb client from Windows 10

closed 2017-05-29 by jchidley

xor-gate, 2017-05-29:
Thank you for your report. Good luck with developing with stlink! Don't hesitate to open a new issue when something is not working.