stlink-org/stlink #600: Cannot write after adress 0x8000800

TheColonelYoung, 2017-06-01:
- [ ] Programmer/board type: Stlink/v2
- [ ] Programmer firmware version: 2.27.15
- [ ] Operating system: Linux - Kubuntu 17.04
- [ ] Stlink tools version : v1.3.1-19-g55c0572
- [ ] Stlink commandline tool name: st-flash
- [ ] Target chip: STM32F072CBT6
- [ ] Git clone : 55c0572, master branch

I try to load my code to chip but when program try to write after 0x8000800 writing is canceled due to error. Code is working because when I use Windows official ST-LINK UTILITY and program the chip all is working fine. Different code on same chip has same result. With this Stlink i programm STMF030C8T6 and all is working. So isn't by programmer or code. 

INPUT:
```
st-flash write DAC.bin 0x08000000
```
OUTPUT
```
st-flash 1.3.1-19-g55c0572
2017-06-01T22:58:35 INFO src/common.c: Loading device parameters....
2017-06-01T22:58:35 INFO src/common.c: Device connected is: F07x device, id 0x20016448
2017-06-01T22:58:35 INFO src/common.c: SRAM size: 0x4000 bytes (16 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 2048 bytes
2017-06-01T22:58:35 INFO src/common.c: Attempting to write 3788 (0xecc) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000800 erased
2017-06-01T22:58:35 INFO src/common.c: Finished erasing 2 pages of 2048 (0x800) bytes
2017-06-01T22:58:35 INFO src/common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2017-06-01T22:58:35 INFO src/flash_loader.c: Successfully loaded flash loader in sram
  0/1 pages written2017-06-01T22:58:35 WARN src/common.c: Failed to unlock flash!
2017-06-01T22:58:39 ERROR src/flash_loader.c: flash loader run error
2017-06-01T22:58:39 ERROR src/common.c: stlink_flash_loader_run(0x8000800) failed! == -1
stlink_fwrite_flash() == -1
```
INPUT:
```
st-flash erase
```
OUTPUT
```
st-flash 1.3.1-19-g55c0572
2017-06-01T22:58:12 INFO src/common.c: Loading device parameters....
2017-06-01T22:58:12 INFO src/common.c: Device connected is: F07x device, id 0x20016448
2017-06-01T22:58:12 INFO src/common.c: SRAM size: 0x4000 bytes (16 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 2048 bytes
Mass erasing
```
Thank you,
TheColonelYoung

xor-gate, 2017-06-10:
It seems the flash loader only erases the flash pages start at addr: 0x08000800. Also the flash loader fails by trying to unlock the flash. Could you try with a little older version from master? Maybe the unlocking is broken for your target.

TheColonelYoung, 2017-06-18:
With older version i got nearly same error but adress is 0x8000000. On other chips works.

Nightwalker-87, 2020-03-21:
@TheColonelYoung: Please retry with Release 1.4.0. This may have been fixed with PR 595.
We need to verify this.

TheColonelYoung, 2020-03-25:
Now I don't have access to specific chip which I used during creating this issue. But I tried to flash another F0 target and it was working correctly. I don't know when (or if) I will be able to test original chip. Tested with releases 1.4.0 and 1.6.0 on STM32F071RB.

Nightwalker-87, 2020-03-25:
The mentioned commit 55c057296ad15fa6de9909206098bd4eed8f6311 is the last before v1.4.0 which targets F0 devices specifically. As you tested with a very similar chip of the same series and verified that v1.4.0 is working, implies that this was not a problem with the tools. As based on the error message and as mentioned before, you seem to have run into a flash unlocking problem regarding the specific chip used, which is very likely not a general problem. Thank you very much for testing. :1st_place_medal: :+1: I'll close this now, as it is not a bug in the tools.

closed 2020-03-25 by Nightwalker-87