stlink-org/stlink #601: STLinkV2 USB protocol, stm8flasher use

janezNov, 2017-06-05:
- [X] Programmer/board type: Stlink/v2
- [X] Operating system: e.g Linux

Please could you point me in direction of RAW USB STLINKV2 command/data frame.
Tool stm8flash:  https://github.com/vdudouyt/stm8flash
stlinkv2.c: https://github.com/vdudouyt/stm8flash/blob/master/stlinkv2.c 
Currently supports only 16bit address and I would like/need to extend it to 24bits, for flash sizes over 32kB

Thank you!
Kind regards

xor-gate, 2017-06-05:
Sorry no support for other tools and repositories, `texane/stlink` is only dedicated to stm32 controllers. Please submit your question to the stm8flash project.

closed 2017-06-05 by xor-gate

janezNov, 2017-06-06:
Thank you on fast respons.
Any suggestions how to sniff this STLINKV2 for SWIM packets as was done for SWD?

xor-gate, 2017-06-06:
Either sniff from a windows virtual machine with VMware (http://esec-lab.sogeti.com/posts/2011/04/06/sniffing-usb-traffic-with-vmware.html) or run windows in a virtualbox virtualmachine on linux and sniff the usb traffic with wireshark.

janezNov, 2017-06-07:
Thank you,
sniffed USB packets while writing 1byte on 0xFFFF/0x10000 address https://github.com/vdudouyt/stm8flash/issues/36
Now need to make sense of it (first by reading existing code)

janezNov, 2017-06-12:
Thank you for suggestion. Solved!