stlink-org/stlink #603: Add support for Nucleo STM32L452

vatsaltrivedi86, 2017-06-09:
**NOTICE: The issue may be closed without notice when not enough information is provided!**

Thank you for giving feedback to the stlink project. Take some time to fill out
 check boxes with a X in the following items so developers and other people can try to
 to find out what is going on. And add/remove what is appropriate to your problem.

When submitting a feature request, try to reuse the list and add/remove what is appropriate.
 Place a `X` between the brackets `[X]` to mark the list item.

- [X] Programmer/board type: Stlink/v2-onboard
- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [X] Operating system: Linux Ubuntu 16.04
- [X] Stlink tools version: v1.3.2
- [X] Stlink commandline tool name: `st-flash`
- [X] Target chip (and optional board):STM32L452RE Nucleo-64 board

When running the st-flash it does not flash the device. It exits after below comments

Output:
user:~/workspace$ ../Packages/stlink.git/build/Release/st-flash --format ihex write STM32L452RE_NUCLEO_RX.hex
st-flash 1.3.1-19-g55c0572-dirty
2017-06-09T08:50:14 INFO src/common.c: Loading device parameters....
2017-06-09T08:50:14 WARN src/common.c: unknown chip id! 0x20016462

Thank you,
The stlink project maintainers

vatsaltrivedi86, 2017-06-10:
I was able to resolve it, I am not sure if my steps are correct but i was able to program the nucleo 64 board. Below are the steps that I flowed to resolve the issue:
 
in file stlink/include/stlink/chipid.h line 60
+	STLINK_CHIPID_STM32_L452 			 = 0x462,

in file stlink/src/chipid.c line 418
+ {
+            // STM32L452RE
+           // From RM0393.
+            .chip_id = STLINK_CHIPID_STM32_L452,
+            .description = "L452RE device",
+            .flash_type = STLINK_FLASH_TYPE_L4,
+            .flash_size_reg = 0x1fff75e0,    // "Flash size data register" (sec 43.2, page 1410)
+            .flash_pagesize = 0x800,         // 2K 
+            .sram_size = 0x28000,
+            .bootrom_base = 0x1fff0000,      // (Bank 1 system memory)
+            .bootrom_size = 0x7000           // 28k (per bank)
+        },

in file stlink/src/flashloader.c line 308 
+ else if ((sl->chip_id == STLINK_CHIPID_STM32_L4) ||
+	       (sl->chip_id == STLINK_CHIPID_STM32_L43X)||
+           (sl->chip_id == STLINK_CHIPID_STM32_L452))

Although i had to run command 
~$ st-flash erase 
before successfully programming the file.

xor-gate, 2017-06-10:
Could you create a Git Pull Request patch for this feature so you get full credits?  More info if you don't know: https://help.github.com/articles/about-pull-requests/

xor-gate, 2017-06-26:
Fixed in #608

PR #608 (2017-06-26): Fix gitignore and add support for STM32L452
Unfortunately, I crammed 2 commits into this PR. The first deals with temporary files generated by `dpkg-buildpackage` on Debian stretch. The second should fix #603.
closed 2017-06-26 by xor-gate