stlink-org/stlink #604: libusb_set_configuration() failed on Mac

mcheung63, 2017-06-11:
Any idea why this happen on Mac?

/Users/peter/Downloads/stlink>./st-flash --reset write ~/Desktop/Nucleo_blink_led_NUCLEO_L476RG.bin 0x8000000
st-flash 1.3.1-19-g55c0572
setting new configuration (0 -> 1)
2017-06-11T20:59:31 WARN src/usb.c: libusb_set_configuration() failed
[!] send_recv send request failed: LIBUSB_ERROR_NOT_FOUND
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_NOT_FOUND
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_NOT_FOUND
[!] send_recv STLINK_DEBUG_ENTER
Failed to enter SWD mode
[!] send_recv send request failed: LIBUSB_ERROR_NOT_FOUND
[!] send_recv STLINK_JTAG_WRITEDEBUG_32BIT

xor-gate, 2017-07-22:
That is odd, it should not happen. Can you give some more info about your setup:

* OS version
* libusb version
* Tried with sudo (last resort)?

mcheung63, 2017-07-25:
i just re-clone the source and compile it again, it works. Anyway here is my information:

/Users/peter>uname -a
Darwin Peter-MacBook-Pro-15.local 16.6.0 Darwin Kernel Version 16.6.0: Fri Apr 14 16:21:16 PDT 2017; root:xnu-3789.60.24~6/RELEASE_X86_64 x86_64

/Users/peter>port list|grep -i usb
usbprog                        @0.2.1_20130915-hg-b833a72 cross/usbprog
libusb                         @1.0.21         devel/libusb
libusb-devel                   @20170716       devel/libusb
libusb-compat                  @0.1.5          devel/libusb-compat
libusb-legacy                  @0.1.12         devel/libusb-legacy
usbmuxd                        @1.1.0          devel/usbmuxd
py-pyusb                       @1.0.0          python/py-pyusb
py27-pyusb                     @1.0.0          python/py-pyusb
py35-pyusb                     @1.0.0          python/py-pyusb
py36-pyusb                     @1.0.0          python/py-pyusb
py-pyusb-devel                 @1.0.0b2        python/py-pyusb-devel
py27-pyusb-devel               @1.0.0b2        python/py-pyusb-devel
py-usbtmc                      @0.8            python/py-usbtmc
py27-usbtmc                    @0.8            python/py-usbtmc
py35-usbtmc                    @0.8            python/py-usbtmc
py36-usbtmc                    @0.8            python/py-usbtmc
usbids                         @2017.02.12     sysutils/usbids
usbutils                       @007            sysutils/usbutils

Thanks

closed 2017-07-25 by mcheung63

xor-gate, 2017-07-25:
Weird, but good it works now. Thank you for your report.