stlink-org/stlink #609: add to documentation: st-flash write is ignoring 0xff at end of file

ponty, 2017-06-29:
- [X] Programmer/board type: `Stlink/v2`
- [ ] Programmer firmware version: 
- [X] Operating system: `Linux Ubuntu 16.04`
- [X] Stlink tools version and/or git commit hash: `v1.3.0-9-g88c6162`
- [X] Stlink commandline tool name: `st-flash`
- [X] Target chip (and optional board): `STM32F415RGT6`

Command:
```
st-flash write firmware.bin 0x8000000
```

Output:

```
Loading device parameters....
Device connected is: F4 device, id 0x10076413
SRAM size: 0x30000 bytes (192 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 16384 bytes
Ignoring 49004 bytes of 0xff at end of file
Attempting to write 213652 (0x34294) bytes to stm32 address: 134217728 (0x8000000)
EraseFlash - Sector:0x0 Size:0x4000 EraseFlash - Sector:0x1 Size:0x4000 EraseFlash - Sector:0x2 Size:0x4000 EraseFlash - Sector:0x3 Size:0x4000 EraseFlash - Sector:0x4 Size:0x10000 EraseFlash - Sector:0x5 Size:0x20000 2017-06-29T09:38:50 INFO /home/gerecsl/aha/workspace/ext/stm32/stlink/src/common.c: Finished erasing 6 pages of 131072 (0x20000) bytes
Starting Flash write for F2/F4/L4
Successfully loaded flash loader in sram
Starting verification of write complete
Flash written and verified! jolly good!
```

After reading the man page I expected that the whole file will be written, 
but the end of the file was ignored because of 0xFF values.
So in this special case an `erase` should be executed first always.
I think that the documentation could have a warning for this.

I tried QSTLink2 for write, and there was no such feature.

xor-gate, 2017-07-22:
Sorry for the late reply, but you could propose a pull request for updating the documentation.

Nightwalker-87, 2020-03-21:
@ponty or anyone else: Please re-test this with Releases v1.4.0 up to v1.6.0 to check if this has been fixed in the meanwhile. If not, please submit a pull request.

Nightwalker-87, 2020-04-09:
As far as I read, EOF (0xff 0xff) is returned by fgetc() due to end of input, so I assume that all further characters beyond are ignored, but I am not sure yet if this is the correct explanation for this behaviour.

I found the following commits that may relate to this:
* 6f194ad388fedf287a6200ea7efb16c4118ca4c1 -> Ignore NULL bytes at flash end when reading
* 379fcb3c614d540e2d854700d77c80268b207b3f -> Ignore ZEROs at end of file when writing
* 6ab0eb83becbe28ea3f97d52b6e5fef39145f08a -> While L1 has empty flash cell erased to zero, at least F1/F4 are erased to 0xff.

chenguokai, 2020-04-10:
As far as I can see, this feature is an optimization.

Assuming we are flashing a STM F1, where after an erase, flash are set to 0xff.

From all over the internet, I found that padding the flash binary file with 0xff is a common practice.

I started from [mikrocontroller.net](mikrocontroller.net), which is mentioned by [https://github.com/stlink-org/stlink/pull/40](https://github.com/stlink-org/stlink/pull/40).

[One post](https://www.mikrocontroller.net/topic/492819#new) reported that Atmel studio filled gaps in hex files.

[This post](https://community.st.com/s/question/0D50X0000Bq9QQO/post-build-armnoneeabiobjcopy-gapfill0xff-in-stm32cubeide) a reply mentioned

> but filling gaps (unused space) in actual Flash with 0xFF is common practice.
This avoids Flash wear, and keeps the programming times low.

[Another post](https://community.nxp.com/thread/453969) also mentioned

> The erased flash will be 0xFF by default no need to rewrite.

This feature would be a bug only when there are problems that flashing violates the following 0xFF default values. I will write a test program to see if it is the case.

If no problems are detected and still there are people who want a precise, expected behavior, shall a new argument be added to stlink for disabling the optimization?

chenguokai, 2020-04-10:
Indeed one bug here. Tested on v1.6.0, stm32f401, with 4KB page size.

If st-flash blindly ignores the ending FF, at least under one circumstance will it cause errors.

Supposing that there is one binary file that ends with consecutive 0xFF. During flash, the addresses that are expected to be filled by 0xFF is ignored. If there was a flash that had wrote some non-FF values, chances are that the non-FF values can be kept.

My proof of concept:

1. Flash one binary file that ends with 0x01010101 (exactly at the boundary of flash page size)
2. Comfirm that the flash is in effect.
3. Flash another binary file that ends with 0xFFFFFFFF at the same boundary.
4. Confirm that 0x01010101 persists.

[poc_binary.zip](https://github.com/stlink-org/stlink/files/4461467/poc_binary.zip)

[image]

Since the optimization is useful in most cases, I suggest adding an arguments for this optimization and disable it by default.

Nightwalker-87, 2020-04-10:
Good, thank you very much for looking into this.

Based on your explanations and finding I'd also consider it a good idea to remove optimisation from the default operation and make it optional by argument. If the latter is then added to our documentation people could then decide just as they wish. Furthermore it is very helpful that you found and confirmed that somehow nasty bug here as well. If you have and idea how to clean this up, you may submit a PR.

Nightwalker-87, 2020-04-10:
Thx for the PR. :+1: Is the feature-request part

> Since the optimization is useful in most cases, I suggest adding an arguments for this optimization and disable it by default.

intended as well on your side? If so, then I'll leave it open for a while.

chenguokai, 2020-04-10:
>  Is the feature-request part intended as well on your side?

In my view, after the coming PR, the documentation shall be updated, providing information about  the current default optimization being disabled and how to reenable if in need.

BTW, I am considering the proper argument name to enable the optimization.
`--optimize-empty-ending` or some other names?

Nightwalker-87, 2020-04-10:
Why not a simple `--opt` together with adding the related definition to the man-page of `st-flash`? I think that's likely the best place for it, as everyone can directly call it via `st-flash --help` then.

chenguokai, 2020-04-10:
I thought there could be more optimizations in the future. If no more optimizations are coming, `--opt` is fair enough.

Nightwalker-87, 2020-04-10:
Yeah, however I wouldn't bother too much about this right now. We'll see how the cookie crumbles. ;-)

chenguokai, 2020-04-11:
Kind of weird behaviour for this optimization:
https://github.com/stlink-org/stlink/blob/99a8aaab253a1cda634291b6a975d1560bb89c0a/src/common.c#L2635

If I got this right, `(num_empty == length) ? (uint32_t) length : (uint32_t) length - num_empty` would flash the whole file when its input is completely empty and four bytes (one word) aligned (rounding down code leads to this requirement)

My current commit keeps this behaviour, while still I think it is not straightforward enough.

Another implementation issue is that I suggest no hint for this optimization on option byte.
Option byte is small enough to prevent any possible optimization and a misconfigured one is really a disaster.

Nightwalker-87, 2020-04-11:
> My current commit keeps this behaviour, while still I think it is not straightforward enough.

Don't be too cautions if things turn out to have not been done nicely. If you are sure about it and feel something is not implemented well, just go for it - we are on a `development` branch and I'll be asking for some reviews and tests on different environments before merging into master for the upcoming release anyway. So different people will look at it.

Nightwalker-87, 2020-04-11:
We can address these minor things at a later point in time when we do some refactoring.
Maybe it's a good idea to add TODOs at suitable places in the code, to ensure this does not go down the sink.

chenguokai, 2020-04-11:
Maybe I can add a TODO tag in the source code itself. 
The latest PR contains only a few lines that describe the feature.

Nightwalker-87, 2020-04-11:
Yes, that would be great.

chenguokai, 2020-04-11:
All should be done now:-)

closed 2020-04-11 by Nightwalker-87