stlink-org/stlink #612: F091RC

binary-nerd, 2017-06-30:
- [X] Programmer/board type: Stlink/v2
- [X] Operating system: Linux
- [X] Stlink commandline tool name: `st-flash` &  `st-util`
- [X] Target chip (and optional board): STM32F091RC

I am seeing the issue mentioned in : https://github.com/texane/stlink/issues/356

I tried mass erase and flash but that did not work.
```
OUTPUT/ERROR of the commandline tool(s)
```
```
F091RC$ st-flash erase
st-flash 1.3.1-19-g55c0572
2017-06-30T18:51:00 INFO src/common.c: Loading device parameters....
2017-06-30T18:51:00 INFO src/common.c: Device connected is: F09X device, id 0x10006442
2017-06-30T18:51:00 INFO src/common.c: SRAM size: 0x8000 bytes (32 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 2048 bytes
Mass erasing

F091RC$ st-flash write ./BUILD/NUCLEO_F091RC/GCC_ARM/non-contact-sensor.bin 0x8000000
st-flash 1.3.1-19-g55c0572
2017-06-30T18:51:11 INFO src/common.c: Loading device parameters....
2017-06-30T18:51:11 INFO src/common.c: Device connected is: F09X device, id 0x10006442
2017-06-30T18:51:11 INFO src/common.c: SRAM size: 0x8000 bytes (32 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 2048 bytes
2017-06-30T18:51:11 INFO src/common.c: Attempting to write 124096 (0x1e4c0) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x0801e000 erased
2017-06-30T18:51:13 INFO src/common.c: Finished erasing 61 pages of 2048 (0x800) bytes
2017-06-30T18:51:13 INFO src/common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2017-06-30T18:51:13 INFO src/flash_loader.c: Successfully loaded flash loader in sram
2017-06-30T18:51:13 ERROR src/flash_loader.c: write error, count == 4294967295
2017-06-30T18:51:13 ERROR src/common.c: stlink_flash_loader_run(0x8000000) failed! == -1
stlink_fwrite_flash() == -1
```

Expected/description:
Should be able to flash without issues

binary-nerd, 2017-07-06:
I could reproduce it again today.

Mass erase did not work.

But when i unplugged st-link from power and connected back(basically reset st-link), i was able to flash..

yohboy, 2017-07-06:
Hi, same problem on  L073RZ.
Mass erase did not work. Reset did not work too.
It works some time but not a lot ...
Always this errors : 
**st-flash 1.4.0-1-g5e740a8
2017-07-06T15:51:16 INFO src/common.c: Loading device parameters....
2017-07-06T15:51:16 INFO src/common.c: Device connected is: L0x Category 5 device, id 0x20086447
2017-07-06T15:51:16 INFO src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x30000 bytes (192 KiB) in pages of 128 bytes
2017-07-06T15:51:16 INFO src/common.c: Attempting to write 105836 (0x19d6c) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08019d00 erased
2017-07-06T15:51:22 INFO src/common.c: Finished erasing 827 pages of 128 (0x80) bytes
2017-07-06T15:51:22 INFO src/common.c: Starting Half page flash write for STM32L core id
2017-07-06T15:51:22 INFO src/flash_loader.c: Successfully loaded flash loader in sram
2017-07-06T15:51:26 ERROR src/flash_loader.c: flash loader run error
2017-07-06T15:51:26 WARN src/common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
2017-07-06T15:51:26 WARN src/common.c: 
write_half_pages failed == -1
825/826 pages written
2017-07-06T15:53:05 INFO src/common.c: Starting verification of write complete
2017-07-06T15:53:07 INFO src/common.c: Flash written and verified! jolly good!**

Of course it's not working.

fuzzyTew, 2017-08-06:
I had this issue, and resolved it by bringing my st-link firmware up-to-date using the windows utility.  It now functions in linux perfectly.

fuzzyTew, 2017-08-06:
Update:  After firmware update, I was able to flash my own code, but I was still unable to flash the bootloader using st-flash or gdb.  The windows utility reported that the pages involved were write-protected in the option bytes.  When told to program the area, it was able to disable the write protection, flash, and then re-enable it, which I suppose is a feature that st-flash lacks.

Nightwalker-87, 2020-03-20:
@binary-nerd @fuzzyTew: Any updates on this? Have you tried with Release v1.4.0?

xloem, 2020-03-20:
I'm sorry, I no longer have ready access to this hardware.

Nightwalker-87, 2020-03-20:
Ok, no problem. Thx for your feedback tough.

Nightwalker-87, 2020-03-31:
Duplicate of #597.

PR #1113 (2021-03-25): Flash loader rework
In this PR, the waiting for data writing is changed from cycles to ms. This should be a more robust solution. 
Fixed a flash loader bug. The flash loader had an incorrect busy wait cycle. The use of bootloader argument registers during boot removed. Removed magic with recalculation of the `size` to the `count`. The number of bytes to load (`size`) is transferred to the flash loader now.
Added disabling of interrupts during flash loader execution. In some cases interrupts caused a failure

(Closes #356, Closes #556, Closes #593, Closes #597, Closes #607, Closes #612, Closes  [... cut]
closed 2021-03-25 by Nightwalker-87