stlink-org/stlink #613: st-util with SWO support possible?

mhanuel26, 2017-07-11:
- [ x] Stlink/v2-onboard
- [x ] Linux
- [ x] `st-util`
- [ x] STM32F769 Discovery

I was wondering if it's possible to have SWO output from target. It will be great to have sort of printf window for debug messages. 
The closest minimalist solution I have found is explained [here](https://stm32f4-discovery.net/2014/12/library-46-debug-stm32f4-device-swo-feature/), but depends on STM32 ST-Link Utility or Keil uVision for it.

Please let me know your thoughts,

xor-gate, 2017-07-11:
Its possible, but not implemented. OpenOCD has support for this to write ITM port 0 or 1 to a file.

shizacat, 2018-01-30:
Hello,
Do you plan to implement swo?

romansavrulin, 2018-03-25:
+1 on SWO feature

langhaarschneider, 2020-04-08:
also +1 on SWO feature

simplerobot, 2021-06-18:
Hi @mhanuel26, @shizacat, @langhaarschneider,
The stlink project now contains a st-trace utility program that will read SWO output from a target.  (Run "st-trace --help" for more details.)  Does this meet your needs?  If not, can you explain how you would like this to work?

Ant-ON, 2021-06-18:
@simplerobot Viewing the SWO output while debugging would be a nice touch. Currently only the semihosting is supported:
https://github.com/stlink-org/stlink/blob/01418b5666c72ce802ea5af88b62dc9aa06c8c3e/src/st-util/gdb-server.c#L1423-L1481

Nightwalker-87, 2022-01-03:
Is there any chance to continue with this?

ChrisIdema, 2022-07-06:
I really need this feature. I see it is added to the list of release v1.7.1. When can I expect the release of v1.7.1?

Nightwalker-87, 2022-07-06:
Currently no date or time frame is planned for any new release.
This is because development is currently mostly on hold within this project, see also: https://github.com/stlink-org/stlink/discussions/1223

Nightwalker-87, 2022-08-27:
I'd consider this a useful feature as well. Who can help out on this topic? Does anybody have an idea how to implement this?

Nightwalker-87, 2024-01-21:
@Ant-ON Would you be interested in taking action here?

ChrisIdema, 2024-01-21:
@stlink-org why did you delete my comment?

linuxguy123, 2024-03-12:
What is the status of SWO working with STLink (st-util) such that Cortex-Debug and uScope (RTT viewer, https://github.com/DEMCON/uscope) can use it ?

If it now does work, does one need to reflash the STLink device to make SWO work or just update st-util ?

linuxguy123, 2024-03-12:
> Hi @mhanuel26, @shizacat, @langhaarschneider, The stlink project now contains a st-trace utility program that will read SWO output from a target. (Run "st-trace --help" for more details.) Does this meet your needs? If not, can you explain how you would like this to work?

Speaking as a VSCode user, how would I configure VSCode to work with st-trace ?  Let's assume I want to use uScope (https://github.com/DEMCON/uscope) as my rtt viewer.

Nightwalker-87, 2024-03-12:
@linuxguy123 There have been no action or any efforts on this topic since the last response of @Ant-ON.
I personally can't contribute to this topic, sorry.

Nightwalker-87, 2024-10-17:
Closed due to inactivity and missing interest.

closed 2024-10-17 by Nightwalker-87