stlink-org/stlink #614: 'No makefile found' error when running make

xtrinch, 2017-07-13:
I'm trying to compile this from source on  Ubuntu 16.04.2 LTS:

test@test:~$ git clone git://github.com/texane/stlink.git && cd stlink

test@test:~/stlink$ ls
build  ChangeLog.md  cmake  CMakeLists.txt  debian  doc  etc  flashloaders  include  LICENSE  Makefile  README.md  scripts  src  stlinkv1_macosx_driver  tests  usr

test@test:~/stlink$ make
[RELEASE]
make[1]: *** No targets specified and no makefile found.  Stop.
Makefile:27: recipe for target 'release' failed

xor-gate, 2017-07-13:
You need to have CMake. Then the toplevel makefile generates makefiles with cmake and starts the build process. It is documented in https://github.com/texane/stlink/blob/master/doc/compiling.md#build-from-sources

xor-gate, 2017-07-13:
Closing this, as it is documented. Feel free to comment if you have further questions.

closed 2017-07-13 by xor-gate

xtrinch, 2017-07-14:
Is there anything other than the package cmake that I need?
```

trinch@marko:~/stlink$ dpkg -s cmake
Package: cmake
Status: install ok installed
Version: 3.5.1-1ubuntu3

```

xor-gate, 2017-07-14:
Yes you need libusb as mentioned in the documentation.

xtrinch, 2017-07-29:
```
trinch@marko:~/stlink$ dpkg -l libusb-1.0*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                                  Version                         Architecture                    Description
+++-=====================================================-===============================-===============================-================================================================================================================
ii  libusb-1.0-0:amd64                                    2:1.0.20-1                      amd64                           userspace USB programming library
ii  libusb-1.0-0:i386                                     2:1.0.20-1                      i386                            userspace USB programming library
ii  libusb-1.0-0-dbg:amd64                                2:1.0.20-1                      amd64                           userspace USB programming library development files
ii  libusb-1.0-0-dev:amd64                                2:1.0.20-1                      amd64                           userspace USB programming library development files
ii  libusb-1.0-doc                                        2:1.0.20-1                      all                             documentation for userspace USB programming
```

I compiled this without using  makefile and I can succesfully use stlink on Ubuntu. It's merely the makefile that isn't working for me.

athirasivaraj, 2018-04-05:
I'm trying to compile this from source on Ubuntu 16.04 LTS

athira@aadi:~/SDx/2017.1/facedetect-fpga-master$ make
sds++ -sds-pf zc706 -sds-hw detectFaces haar.cpp -clkid 3 -sds-end -poll-mode 1   -Wall -O3 -c -MMD -MP -MF"image.d" image.cpp -o image.o > sdsoc.log
/bin/sh: 1: sds++: not found
Makefile:27: recipe for target 'image.o' failed
make: *** [image.o] Error 127
ANY IDEAS

Vascom, 2018-04-05:
Add path to sds++ to your PATH.

slyshykO, 2018-04-05:
@athirasivaraj , @Vascom . Is it still about stlink?

Vascom, 2018-04-05:
@slyshykO I don't see command first time. Seems that it is not stlink :)

athirasivaraj, 2018-04-05:
I have done with this export path command, but still its not working. So any other ideas in adding path of sds++ to my project directory path. 
I am attaching the screenshot
[image]

Vascom, 2018-04-05:
Seems that it is https://en.wikipedia.org/wiki/Internet_troll

slyshykO, 2018-04-05:
Dear @athirasivaraj  , please, write to a proper project (https://github.com/cornell-zhang/facedetect-fpga)

xor-gate, 2018-04-06:
@athirasivaraj screaming for free help on the internet is not going to help you, probably you are missing some needed linux knowledge. Good luck.

Nightwalker-87, 2020-03-17:
Issue closed as invalid and not related to the st-link project.