stlink-org/stlink #616: ERROR src/gdbserver/gdb-server.c: cannot check for int: -2

JustAnother1, 2017-07-15:
- [X] Programmer/board type: Stlink/v2-onboard
- [X] Programmer firmware version: V2.J27.M15
- [X] Operating system: Linux (Mint )  4.8.0-58-generic #63~16.04.1-Ubuntu SMP Mon Jun 26 18:08:51 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
- [X] Stlink tools version and/or git commit hash: st-util 1.4.0-1-g5e740a8
- [X] Stlink commandline tool name: `st-util`
- [X] Target chip (and optional board): e.g STM32F401RET6U Nucleo-F401RE

I compiled my code and flashed the elf file. I then started st-util and connected to it with Eclipse (GDB). It stopped at main I then tried to single step (F5) but after a few steps st-util died.
My firmware is probably doing something completely stupid, but that is the reason for starting a GDB session,..
I have this issue with different versions of my firmware. Sometime reflashing the same code made it go away.

Output:

```
02:25 lars@merlin ~/git/stlink/build/src/gdbserver $ ./st-util -v99
st-util 1.4.0-1-g5e740a8
2017-07-15T02:26:46 DEBUG src/common.c: stlink current mode: mass
2017-07-15T02:26:46 DEBUG src/common.c: stlink current mode: mass
2017-07-15T02:26:46 DEBUG src/common.c: *** stlink_enter_swd_mode ***
2017-07-15T02:26:46 DEBUG src/common.c: *** looking up stlink version
2017-07-15T02:26:46 DEBUG src/common.c: st vid         = 0x0483 (expect 0x0483)
2017-07-15T02:26:46 DEBUG src/common.c: stlink pid     = 0x374b
2017-07-15T02:26:46 DEBUG src/common.c: stlink version = 0x2
2017-07-15T02:26:46 DEBUG src/common.c: jtag version   = 0x1b
2017-07-15T02:26:46 DEBUG src/common.c: swim version   = 0xf
2017-07-15T02:26:46 DEBUG src/common.c: *** stlink_jtag_reset ***
2017-07-15T02:26:46 DEBUG src/common.c: *** stlink_reset ***
2017-07-15T02:26:46 INFO src/common.c: Loading device parameters....
2017-07-15T02:26:46 DEBUG src/common.c: *** stlink_core_id ***
2017-07-15T02:26:46 DEBUG src/common.c: core_id = 0x2ba01477
2017-07-15T02:26:46 DEBUG src/common.c: *** stlink_read_debug32 10006433 is 0xe0042000
2017-07-15T02:26:46 DEBUG src/common.c: *** stlink_read_debug32 200c000 is 0x1fff7a20
2017-07-15T02:26:46 INFO src/common.c: Device connected is: F4 device (Dynamic Efficency), id 0x10006433
2017-07-15T02:26:46 INFO src/common.c: SRAM size: 0x18000 bytes (96 KiB), Flash: 0x80000 bytes (512 KiB) in pages of 16384 bytes
2017-07-15T02:26:46 DEBUG src/common.c: *** set_swdclk ***
2017-07-15T02:26:46 DEBUG src/common.c: *** stlink_reset ***
2017-07-15T02:26:46 INFO src/gdbserver/gdb-server.c: Chip ID is 00000433, Core ID is  2ba01477.
2017-07-15T02:26:46 INFO src/gdbserver/gdb-server.c: Listening at *:4242...
2017-07-15T02:26:51 DEBUG src/common.c: *** stlink_force_debug_mode ***
2017-07-15T02:26:51 DEBUG src/common.c: *** stlink_reset ***
2017-07-15T02:26:51 DEBUG src/common.c: *** stlink_write_debug32 3 to 0xe0002000
2017-07-15T02:26:51 DEBUG src/common.c: *** stlink_read_debug32 261 is 0xe0002000
2017-07-15T02:26:51 INFO src/gdbserver/gdb-server.c: Found 6 hw breakpoint registers
2017-07-15T02:26:51 DEBUG src/common.c: *** stlink_write_debug32 0 to 0xe0002008
2017-07-15T02:26:51 DEBUG src/common.c: *** stlink_write_debug32 0 to 0xe000200c
2017-07-15T02:26:51 DEBUG src/common.c: *** stlink_write_debug32 0 to 0xe0002010
2017-07-15T02:26:51 DEBUG src/common.c: *** stlink_write_debug32 0 to 0xe0002014
2017-07-15T02:26:51 DEBUG src/common.c: *** stlink_write_debug32 0 to 0xe0002018
2017-07-15T02:26:51 DEBUG src/common.c: *** stlink_write_debug32 0 to 0xe000201c
2017-07-15T02:26:51 DEBUG src/gdbserver/gdb-server.c: init watchpoints
2017-07-15T02:26:51 DEBUG src/common.c: *** stlink_read_debug32 1 is 0xe000edfc
2017-07-15T02:26:51 DEBUG src/common.c: *** stlink_write_debug32 1000001 to 0xe000edfc
2017-07-15T02:26:51 DEBUG src/common.c: *** stlink_write_debug32 0 to 0xe0001028
2017-07-15T02:26:51 DEBUG src/common.c: *** stlink_write_debug32 0 to 0xe0001038
2017-07-15T02:26:51 DEBUG src/common.c: *** stlink_write_debug32 0 to 0xe0001048
2017-07-15T02:26:51 DEBUG src/common.c: *** stlink_write_debug32 0 to 0xe0001058
2017-07-15T02:26:51 INFO src/gdbserver/gdb-server.c: GDB connected.
2017-07-15T02:26:51 DEBUG src/gdbserver/gdb-server.c: recv: qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+
2017-07-15T02:26:51 DEBUG src/gdbserver/gdb-server.c: query: Supported;multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+
2017-07-15T02:26:51 DEBUG src/gdbserver/gdb-server.c: send: PacketSize=3fff;qXfer:memory-map:read+
2017-07-15T02:26:51 DEBUG src/gdbserver/gdb-server.c: recv: vMustReplyEmpty
2017-07-15T02:26:51 DEBUG src/gdbserver/gdb-server.c: send: 
2017-07-15T02:26:51 DEBUG src/gdbserver/gdb-server.c: recv: Hg0
2017-07-15T02:26:51 DEBUG src/gdbserver/gdb-server.c: send: 
2017-07-15T02:26:51 DEBUG src/gdbserver/gdb-server.c: recv: qTStatus
2017-07-15T02:26:51 DEBUG src/gdbserver/gdb-server.c: query: TStatus;
2017-07-15T02:26:51 DEBUG src/gdbserver/gdb-server.c: send: 
2017-07-15T02:26:51 DEBUG src/gdbserver/gdb-server.c: recv: ?
2017-07-15T02:26:51 DEBUG src/gdbserver/gdb-server.c: send: S05
2017-07-15T02:26:51 DEBUG src/gdbserver/gdb-server.c: recv: qfThreadInfo
[... 133 lines cut ...]
2017-07-15T02:26:53 DEBUG src/common.c: *** stlink_write_debug32 48006e21 to 0xe0002010
2017-07-15T02:26:53 DEBUG src/gdbserver/gdb-server.c: send: OK
2017-07-15T02:26:53 DEBUG src/gdbserver/gdb-server.c: recv: Z1,8006e26,2
2017-07-15T02:26:53 DEBUG src/gdbserver/gdb-server.c: setting hw break 3 at 08006e24 (2)
2017-07-15T02:26:53 DEBUG src/gdbserver/gdb-server.c: reg 88006e25 
2017-07-15T02:26:53 DEBUG src/common.c: *** stlink_write_debug32 88006e25 to 0xe0002014
2017-07-15T02:26:53 DEBUG src/gdbserver/gdb-server.c: send: OK
2017-07-15T02:26:53 DEBUG src/gdbserver/gdb-server.c: recv: vCont?
2017-07-15T02:26:53 DEBUG src/gdbserver/gdb-server.c: send: 
2017-07-15T02:26:53 DEBUG src/gdbserver/gdb-server.c: recv: Hc0
2017-07-15T02:26:53 DEBUG src/gdbserver/gdb-server.c: send: 
2017-07-15T02:26:53 DEBUG src/gdbserver/gdb-server.c: recv: c
2017-07-15T02:26:53 DEBUG src/common.c: *** stlink_run ***
2017-07-15T02:26:53 DEBUG src/common.c: *** stlink_status ***
2017-07-15T02:26:53 DEBUG src/common.c:   core status: running
2017-07-15T02:26:53 ERROR src/gdbserver/gdb-server.c: cannot check for int: -2
2017-07-15T02:26:54 DEBUG src/common.c: *** stlink_run ***
2017-07-15T02:26:54 DEBUG src/common.c: *** stlink_exit_debug_mode ***
2017-07-15T02:26:54 DEBUG src/common.c: *** stlink_write_debug32 a05f0000 to 0xe000edf0
2017-07-15T02:26:54 DEBUG src/common.c: *** stlink_close ***
02:26 lars@merlin ~/git/stlink/build/src/gdbserver $ 
```

Expected/description:
I expected to be able to continue debugging.

xor-gate, 2017-07-22:
Thats odd, i'm not sure whats wrong with the gdb-server stub. Does `st-flash` work?

JustAnother1, 2017-07-22:
Yes st-flash works.

JustAnother1, 2017-07-31:
The Nucleo I used was rather old. (It did come with X2 not populated and the type sticker on ST-link part of the board). I think the STM32F401RE (Main processor on the board) has been damaged and does not work completely according to spec anymore.
I bought a new Nucleo (X2 is populated, type sticker is next to main cpu) and I could debug on the new board. I since have been able to get to a firmware that works on the new Nucleo. The same firmware does not work on the old Nucleo. Therefore I assume that parts of the old Nucleo processor are damaged.

I also assume that this error only happens on such partly damaged devices. This makes this bug less severe. It might still be a good Idea to make the debug messages more clear.

Thanks for looking into this.

Nightwalker-87, 2020-07-18:
> It might still be a good Idea to make the debug messages more clear.

True.

Has anyone else experienced something similar with the STM32F401RET6U / Nucleo-F401RE?

Nightwalker-87, 2021-03-13:
If the used hardware was at least partly defective, one may assume that this also led to the suspicious output.

If functionality of a MCU is in doubt and strange behaviour occurs at the same time, it is often a good advice to simply dump the related hardware (especially if not very expensive) and to start over again with a replacement. In most cases it is unlikely that one is able to allocate the exact cause of failure.

As the initially observed behaviour could not be reproduced with a working board, we can safely conclude that there was no bug in the toolset in this case. Improvement of debug messages should have seen some improvements in the meanwhile.

closed 2021-03-13 by Nightwalker-87