stlink-org/stlink #623: Windows st-util could not retrieve port number for device

xtrinch, 2017-07-29:
Trying to run st-util on windows 10 for connected board STM32F769I-DISCO:

```
C:\Users\trinch> st-info --probe
Found 1 stlink programmers
```

```
PS C:\Users\trinch> st-info --chipid
libusb: warning [windows_get_device_list] could not retrieve port number for device '\\.\USBESTUB#DEVICES#0004', skipping: [13] The data is invalid.
libusb: warning [windows_get_device_list] could not retrieve port number for device '\\.\USBESTUB#DEVICES#0000', skipping: [13] The data is invalid.
0x0451
```

```
C:\Users\trinch> st-util
STLINK GDB Server v0.5.6 (Mar 24 2013 10:29:19)
Many thanks to the STLINK development team.
(https://github.com/texane/stlink)

2017-07-29T17:48:37 WARN src/stlink-usb.c: Couldn't find any ST-Link/V2 devices
```

Though, I can program my device with ST-LINK_CLI. Help?

Nightwalker-87, 2020-02-23:
@xtrinch which libusb version is used? Please also test this with Release v1.4.0, v1.5.0, v1.5.1 and v.1.6.0, which have been released in the meanwhile, as this issue is already very old.

Nightwalker-87, 2020-03-22:
ping @xtrinch

slyshykO, 2020-03-22:
@Nightwalker-87, Think it is connected with #438 also.

Nightwalker-87, 2020-03-23:
> ```
> PS C:\Users\trinch> st-info --chipid
> libusb: warning [windows_get_device_list] could not retrieve port number for device '\\.\USBESTUB#DEVICES#0004', skipping: [13] The data is invalid.
> libusb: warning [windows_get_device_list] could not retrieve port number for device '\\.\USBESTUB#DEVICES#0000', skipping: [13] The data is invalid.
> 0x0451
> 
> PS C:\Users\trinch> st-info --descr
> libusb: warning [windows_get_device_list] could not retrieve port number for device '\\.\USBESTUB#DEVICES#0004', skipping: [13] The data is invalid.
> libusb: warning [windows_get_device_list] could not retrieve port number for device '\\.\USBESTUB#DEVICES#0000', skipping: [13] The data is invalid.
> F76xxx device
> ```

@slyshykO: Yes, this obviously is the same as in your MinGW test there.

Duplicate of #438.

xtrinch, 2020-03-24:
Can't offer feedback which version of libusb it was as I don't have the setup anymore

Nightwalker-87, 2020-03-24:
@xtrinch: That was expected due to the time passed. We're fine with this.
I think it will not affect us here as we also have #438.

closed 2020-04-10 by Nightwalker-87