stlink-org/stlink #625: make release error

unlimited123az, 2017-08-08:
I saw a similar issue,but I have installed cmake,libusb,gcc
I try to run "make release" ,however
[RELEASE]
make[1]: *** No targets specified and no makefile found.  Stop.
Makefile:27: recipe for target 'release' failed
make: *** [release] Error 2

I have no idea.

I tried "make debug"
It stops halfway.
I tried again,and then
[DEBUG]
make[1]: *** No targets specified and no makefile found.  Stop.
Makefile:23: recipe for target 'debug' failed
make: *** [debug] Error 2

xor-gate, 2017-08-08:
First run `make clean` after cmake fails then run `make` again.

unlimited123az, 2017-08-09:
Many thanks，it works well！

unlimited123az, 2017-08-09:
One more question............
I plug in my stm32 board,but 
/dev/stlink dose not exist,I can not execute 
st-util -1

xor-gate, 2017-08-09:
`/dev/stlink` is only created by a custom udev rule. Probably you should run `st-util` without arguments.

unlimited123az, 2017-08-09:
There is no st-uil 
bash:st-uil:command not found

xor-gate, 2017-08-09:
The `st-util` is there where it is build, see https://github.com/texane/stlink/blob/master/doc/compiling.md.

unlimited123az, 2017-08-11:
Many thanks,
I can keep on learning stm32 now.

closed 2017-08-11 by unlimited123az