stlink-org/stlink #626: [NUCLEO] stm32l053r8 - unknown chip id! 0x10086417

OBU5, 2017-08-09:
Hello guys. 

I have Problem with uploading binary to Board

I have Nucleo board stm32l053r8, and I am able to upload there binary with STSW-LINK004.
But It is for Windows and I am programming on Linux. When I was uploading binary to Discovery board, there was no Problem, but now, when I wrote in bash "st-flash erase" or , bash told me "unknown chip id! 0x10086417"
(like in the first picture)
[image]

When I write it with my discovery board (STM32 F0308DISCOVERY), everything seems alright.
(like in the second picture)
[image]

dont you know, where should be a problem? :( ...

Please help me :( I am lost

src/stlink-common.c: unknown chip id! 0x10086417

xor-gate, 2017-08-09:
Please provide the requested information (as per [ISSUE_TEMPLATE.md](https://github.com/texane/stlink/blob/master/.github/ISSUE_TEMPLATE.md)) about which st-util you are using. Probably your version is to old. Since v1.2.0 L053 is supported (see [ChangeLog.md](https://github.com/texane/stlink/blob/master/ChangeLog.md#v120))

OBU5, 2017-08-09:
Thank you for your reply.

Yeah I realy had a old version. Thank you very much !!!!! :)

xor-gate, 2017-08-09:
Great, thank you for your reply.

closed 2017-08-09 by xor-gate