stlink-org/stlink #627: gdb hangs with "step" and cannot be unhang with ^C

SQLek, 2017-08-16:
I'm trying do debug hanging up procesor/debugger.

Output of st-util on verbosity 99 is:
```
2017-08-17T00:04:14 DEBUG src/gdbserver/gdb-server.c: send: 0000004000000000000000000000000000200240c28ba50053043f090140a962ed150008ffffffffd9d6b36426249c5900003210f84f0020bb1000085a110008
2017-08-17T00:04:14 DEBUG src/gdbserver/gdb-server.c: recv: m800115a,4
2017-08-17T00:04:14 DEBUG src/common.c: *** stlink_read_mem32 ***
2017-08-17T00:04:14 DEBUG src/sg.c: Sending usb m-s cmd: cdblen:10, rxsize=8
2017-08-17T00:04:14 DEBUG src/sg.c: Sending CDB [ 0xf2 0x7 0x58 0x11 00 0x8 0x8 00 00 00]
2017-08-17T00:04:14 DEBUG src/gdbserver/gdb-server.c: send: fee700b5
2017-08-17T00:04:14 DEBUG src/gdbserver/gdb-server.c: recv: s
2017-08-17T00:04:14 DEBUG src/common.c: *** stlink_step ***
2017-08-17T00:04:14 DEBUG src/sg.c: Sending usb m-s cmd: cdblen:10, rxsize=2
2017-08-17T00:04:14 DEBUG src/sg.c: Sending CDB [ 0xf2 0xa 00 00 00 00 00 00 00 00]
2017-08-17T00:04:14 DEBUG src/sg.c:   step core: ok
2017-08-17T00:04:14 DEBUG src/gdbserver/gdb-server.c: send: S05
2017-08-17T00:04:14 DEBUG src/gdbserver/gdb-server.c: recv: g
2017-08-17T00:04:14 DEBUG src/common.c: *** stlink_read_all_regs ***
2017-08-17T00:04:14 DEBUG src/sg.c: Sending usb m-s cmd: cdblen:10, rxsize=84
2017-08-17T00:04:14 DEBUG src/sg.c: Sending CDB [ 0xf2 0x4 00 00 00 00 00 00 00 00]
2017-08-17T00:04:14 DEBUG src/gdbserver/gdb-server.c: send: 0000004000000000000000000000000000200240c28ba50053043f090140a962ed150008ffffffffd9d6b36426249c5900003210f84f0020bb1000085a110008
```
all over, in repetition.

But gdb (arm-none-eabi-gdb) is hang, and i cannot unhang it with ^C.

Place in code doesn't seem to make any sense. Touching anything with interrupts cause strange behavior. I was able to do TIM4 interrupts but enabling USART3 without interrupt enabled, still triger hangs.

In most cases offending function is MX_NVIC_Init();

I would like to debug problem, but i can't because in this stage gdb is dead/hang. Can be unhang with killing st-util. or hard kill from system.

Programer: StLink V1 orginal
CPU: Stm32f103c8 on arduino-mini-style-board
Link type: SWD
OS: `Linux sqlek 4.11.0-1-amd64 #1 SMP Debian 4.11.6-1 (2017-06-19) x86_64 GNU/Linux`
App name/version: st-util/v1.4.0 downloaded 5 days ago.
Firmware version: how to check?
Also Stm32CubeMX latest.

Any idea how to proceed?

- [x ] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [x ] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
- [x ] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
- [x ] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util`
- [x ] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

xor-gate, 2017-08-17:
I'm not sure what causes the problem, stlink v1 programmers are not widely used anymore and may cause problem. You are the first one reporting this kind of problem.

SQLek, 2017-08-17:
In my country, considering time, i had option to buy original v1 or china v2.

Tomorrow i'll buy this http://www.aprovi.com.pl/zdjecia/ply155.jpg

Problem can be also with gdb:
Gdb version: GNU gdb (7.12-6+9+b2) 7.12.0.20161007-git
Firmware version was latest V1.J13.S4, but i reflashed it anyway.

In weekend i'm doing more investigation.

Nightwalker-87, 2020-03-19:
@SQLek: Please verify if the problem still exists in Release v1.5.0.

Nightwalker-87, 2020-07-18:
We should know if the problem still exists on our `v1.5.0-patch` branch. Can somebody contribute?

Nightwalker-87, 2021-04-04:
This appears to be a `gdb`-related issue which is not related to the `stlink` toolset.

Further explanation can be found here: https://undo.io/resources/gdb-watchpoint/how-work-signals-gdb/

The reference above describes the case of `gdb` not responding to CTRL+C.
Thus we can now safely close this issue as resolved.

closed 2021-04-04 by Nightwalker-87