stlink-org/stlink #628: clang 4.0.0+ can not compile version 1.4.0

blacklion, 2017-08-21:
clang 4.0.0 or newer can not compile 1.4.0 release because it uses
```
add_cflag_if_supported("-Wall")
add_cflag_if_supported("-Wextra")
add_cflag_if_supported("-Werror")
```
in `CFlags.cmake`, and later add integer to string literal (in logging code), which is warning for clang. Build log is attached,

[build.txt](https://github.com/texane/stlink/files/1239240/build.txt)

xor-gate, 2017-08-21:
This is the same error as build with ninja and GCC in PR #624. We need to fix this in the logging module.

slyshykO, 2017-10-25:
we can use strrchr ( [](http://en.cppreference.com/w/c/string/byte/strrchr) ) for extracting the base name.
Another problem is the dir separator on windows depends on compiler and cmake generator. And can be both '\' and '/' .

xor-gate, 2017-10-25:
Yes for windows you can have multiple path separators (cygwin/mingw/native).

slyshykO, 2017-10-29:
Is it still a bug?

blacklion, 2017-10-29:
I've checked very last "master" (95b6e03ed0c0ee40256a33f04eedf702005a9d8b), and it build flawlessly with clang 5.0.0 on 64-bit system.
Thank you!

xor-gate, 2017-10-29:
Thanks all for being proactive.

closed 2020-03-18 by Nightwalker-87