stlink-org/stlink #632: [Question] : st-info --probe not listing device info

studavekar, 2017-08-31:
The motive here is 

1. Flash the device using st-link, instead of drag-drop method supported by mbed enabled device (issue are seen the ST device doesn't get programmed always with drag drop)
2. Flashing multiple devices with same binary. 
3. How hard reset device if possible. 
```
➜  stlink git:(master) ✗ sudo  build/Release/st-info --probe
Found 5 stlink programmers
```

```
➜  stlink git:(master) ✗ sudo  build/Release/st-info --serial
303637314646343934393531373835

```
Device connected , listing by mbed-ls all device are mbed enabled 
```
| NUCLEO_F401RE | NUCLEO_F401RE[0]     | /mnt/pci-0000_00_14_0-usb-0_1_2_1_1-scsi-0_0_0_0  | /dev/ttyACM1  | 07200221072265023866F31F                         | 0221            |
| NUCLEO_F401RE | NUCLEO_F401RE[1]     | /mnt/pci-0000_00_14_0-usb-0_1_3_1_1-scsi-0_0_0_0  | /dev/ttyACM3  | 07200221073265613876F37C                         | 0221            |
| NUCLEO_F401RE | NUCLEO_F401RE[2]     | /mnt/pci-0000_00_14_0-usb-0_1_1_1_1-scsi-0_0_0_0  | /dev/ttyACM0  | 072002210774650F3830F312                         | 0221            |
| NUCLEO_F429ZI | NUCLEO_F429ZI[0]     | /mnt/pci-0000_00_14_0-usb-0_6_2_1_1-scsi-0_0_0_0  | /dev/ttyACM5  | 07960221004C62643F08F4CF                         | 0221            |
| NUCLEO_F429ZI | NUCLEO_F429ZI[1]     | /mnt/pci-0000_00_14_0-usb-0_6_3_1_1-scsi-0_0_0_0  | /dev/ttyACM8  | 07960221012567073E61F1AC                         | 0221            |
+---------------+----------------------+---------------------------------------------------+---------------+--------------------------------------------------+-----------------+
```

xor-gate, 2017-08-31:
1. The mbed protocol is different than the stlink protocol. It is a current limitation with probe it will not show the programmer serial numbers when no target is connected. I have rewritten a part of stlink for myself which makes probe work without a connected target with my [stlink2 project](https://github.com/xor-gate/stlink2).
2. Flashing multiple targets at once with the same binary is not possible with the current st-flash tool, you could write a shell script to make this possible
3. Hard reset, thats a good question. There are two resets, the pin reset and the core reset over the programming interface. I can not answer which is implemented for texane/stlink (see also [#518](https://github.com/texane/stlink/issues/518#issuecomment-272627708).

You are also able to write your own tool around the stlink library, as it is build as an shared object (dll). But also can be staticly linked. Have a look at how simple the st-flash tool is [src/tools/flash.c](https://github.com/texane/stlink/blob/master/src/tools/flash.c)

studavekar, 2017-08-31:
Thanks @xor-gate

closed 2017-08-31 by studavekar