stlink-org/stlink #634: stlink library calls exit() or _exit()

Vascom, 2017-09-04:
When I am build package stlink for Fedora it make some ckecks. And I see this warning:
stlink.x86_64: W: shared-lib-calls-exit /usr/lib64/libstlink-shared.so.1.4.0 exit@GLIBC_2.2.5

shared-lib-calls-exit:
This library package calls exit() or _exit(), probably in a non-fork()
context. Doing so from a library is strongly discouraged - when a library
function calls exit(), it prevents the calling program from handling the
error, reporting it to the user, closing files properly, and cleaning up any
state that the program has. It is preferred for the library to return an
actual error code and let the calling program decide how to handle the
situation.

Can you correct it?

xor-gate, 2017-09-04:
I totaly agree. Exit is evil for libraries (`.so`, `.a`). Feel free to submit a PR, i'm fairly busy these days.

Vascom, 2017-09-08:
PR #636 created.

commit 7ec1fb78 (2018-03-28): remove exit() from lib. fix #634 (#696)

* remove exit() from lib. fix #634
* remove the UFATAL loglevel completely.
closed 2020-03-18 by Nightwalker-87