stlink-org/stlink #638: STM32F4 not flashing

banuprathap, 2017-09-28:
Hi,
I am unable to flash the compiled file. I am using **STM32F407IGT6** on a custom board. It shows `stlink_fwrite_flash() == -1` error. I tried reinstalling stlink but to no avail.

- [X] Programmer/board type: Stlink/v2
- [X] Operating system: 4.4.0-96-generic #119~14.04.1-Ubuntu SMP i686 GNU/Linux
- [X] Stlink tools version and/or git commit hash: 1.4.0-9-g6511a7e
- [X] Stlink commandline tool name:  `st-flash`
- [X] Target chip (and optional board): STM32F407IGT6
A as-detailed description possible of the problem with debug output when available.

Output:

```
st-flash write build/flash.enc 0x8010000
st-flash 1.4.0-9-g6511a7e
2017-09-28T12:43:48 INFO src/common.c: Loading device parameters....
2017-09-28T12:43:48 INFO src/common.c: Device connected is: F4 device, id 0x10036413
2017-09-28T12:43:48 INFO src/common.c: SRAM size: 0x30000 bytes (192 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 16384 bytes
2017-09-28T12:43:48 INFO src/common.c: Attempting to write 257772 (0x3eeec) bytes to stm32 address: 134283264 (0x8010000)
Flash page at addr: 0x08040000 erasedEraseFlash - Sector:0x6 Size:0x20000 
2017-09-28T12:43:54 INFO src/common.c: Finished erasing 3 pages of 131072 (0x20000) bytes
2017-09-28T12:43:54 INFO src/common.c: Starting Flash write for F2/F4/L4
2017-09-28T12:43:54 INFO src/flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 32768
2017-09-28T12:44:04 ERROR src/flash_loader.c: flash loader run error
2017-09-28T12:44:04 ERROR src/common.c: stlink_flash_loader_run(0x8010000) failed! == -1
stlink_fwrite_flash() == -1
```
It was working fine couple of days ago but wouldn't work anymore. Am I doing anything wrong?

Thank you!

xor-gate, 2017-09-29:
Flash loader problems are hard to debug. Maybe you can find some usefull information from other open and closed issues by filtering: https://github.com/texane/stlink/labels/error%2Fflash-loader.

fernandolopez, 2018-03-08:
I had a similar problem with a STM32F446RET6 (writing at address 0x8000000). After a mass erase it started to work.

Nightwalker-87, 2020-02-21:
Please verify if problem still exists in Release v1.6.0.

banuprathap, 2020-02-27:
I tried v1.6.0 as I couldn't find v1.6.1 and it seems to work now. Thank you!

Nightwalker-87, 2020-02-27:
v1.6.1 is only the milestone of the next upcoming release. Nothing to worry about. You did everything right! Thank you for your feedback - very helpful to us. :-)

Nightwalker-87, 2020-02-27:
Feedback resolves #638.

Nightwalker-87, 2020-04-06:
Reopening this as the underlying problem that causes this issue remains unsolved.

PR #1113 (2021-03-25): Flash loader rework
In this PR, the waiting for data writing is changed from cycles to ms. This should be a more robust solution. 
Fixed a flash loader bug. The flash loader had an incorrect busy wait cycle. The use of bootloader argument registers during boot removed. Removed magic with recalculation of the `size` to the `count`. The number of bytes to load (`size`) is transferred to the flash loader now.
Added disabling of interrupts during flash loader execution. In some cases interrupts caused a failure

(Closes #356, Closes #556, Closes #593, Closes #597, Closes #607, Closes #612, Closes  [... cut]
closed 2021-03-25 by Nightwalker-87