stlink-org/stlink #640: STM32L476RG - Compilation and running are good but there is no green light in STLink module

remi22, 2017-09-29:
Hi,
I succeed in compiling and running (INFO src/common.c: Flash written and verified! jolly good!) but there is still a red light in the STLink module and the LED2 doesn't blink. I would like to know if you have tried this method on STM32L476RG.

`2017-09-29T13:07:42 INFO src/common.c: Loading device parameters....
2017-09-29T13:07:42 INFO src/common.c: Device connected is: L4 device, id 0x10076415
2017-09-29T13:07:42 INFO src/common.c: SRAM size: 0x18000 bytes (96 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 2048 bytes
2017-09-29T13:07:42 INFO src/common.c: Attempting to write 3484 (0xd9c) bytes to stm32 address: 134217728 (0x8000000)
EraseFlash - Page:0x0 Size:0x800 st-flash 1.4.0

Flash page at addr: 0x08000000 erasedEraseFlash - Page:0x1 Size:0x800 2017-09-29T13:07:42 
Flash page at addr: 0x08000800 erasedINFO src/common.c: Finished erasing 2 pages of 2048 (0x800) bytes
2017-09-29T13:07:42 INFO src/common.c: Starting Flash write for F2/F4/L4
2017-09-29T13:07:42 INFO src/flash_loader.c: Successfully loaded flash loader in sram
2017-09-29T13:07:42 INFO src/common.c: Starting verification of write complete
2017-09-29T13:07:42 INFO src/common.c: Flash written and verified! jolly good!

size: 3484
`

Thank you,
BICHE Rémi

xor-gate, 2017-10-10:
I'm not sure but probably your application is not running after flashing (reset is not performed after flash). I think you mean LED2 on your board or on the programmer?

Nightwalker-87, 2020-03-19:
@remi22: What is the current state of this? Have you found a solution?

remi22, 2020-03-19:
Hi, I am sorry to answer late. It's been quite a while since I used an STM32. I really don't remember why I had the issue.

closed 2020-03-19 by remi22