stlink-org/stlink #642: Anoder solution #356, --reset parameter

MarekLew, 2017-10-12:
- [ ] Programmer/board type: Stlink/v2
- [ ] Programmer firmware version: e.g STM32 STLink 2
- [ ] Operating system: Raspbian GNU/Linux 8.0 (jessie)
- [ ] Stlink tools version and/or git commit hash: 1.4.0-11-ga2a707e
- [ ] Stlink commandline tool name: e.g st-flash
- [ ] Target chip (and optional board): e.g STM32F030C8

When I stared, I have  problem #356. Problem occurs second or third upload. 
Output:

```
user@raspberrypi:~/apps/stm_soft $ st-flash --format ihex write Test.hex
st-flash 1.4.0-11-ga2a707e
2017-10-12T14:16:53 INFO src/common.c: Loading device parameters....
2017-10-12T14:16:53 INFO src/common.c: Device connected is: F0 device, id 0x20006440
2017-10-12T14:16:53 INFO src/common.c: SRAM size: 0x2000 bytes (8 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2017-10-12T14:16:53 INFO src/common.c: Attempting to write 19320 (0x4b78) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08004800 erased
2017-10-12T14:16:53 INFO src/common.c: Finished erasing 19 pages of 1024 (0x400) bytes
2017-10-12T14:16:53 INFO src/common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2017-10-12T14:16:53 INFO src/flash_loader.c: Successfully loaded flash loader in sram
2017-10-12T14:16:58 ERROR src/flash_loader.c: flash loader run error
2017-10-12T14:16:58 ERROR src/common.c: stlink_flash_loader_run(0x8000000) failed! == -1
stlink_fwrite_flash() == -1

```
After that I erase the flash and it`s work good.

Then I use --reset parameter and this work perfect. Works every time.
```
user@raspberrypi:~/apps/stm_soft $ st-flash --reset --format ihex write Test.hex
st-flash 1.4.0-11-ga2a707e
2017-10-12T14:25:35 INFO src/common.c: Loading device parameters....
2017-10-12T14:25:35 INFO src/common.c: Device connected is: F0 device, id 0x20006440
2017-10-12T14:25:35 INFO src/common.c: SRAM size: 0x2000 bytes (8 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2017-10-12T14:25:35 INFO src/common.c: Attempting to write 19320 (0x4b78) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08004800 erased
2017-10-12T14:25:36 INFO src/common.c: Finished erasing 19 pages of 1024 (0x400) bytes
2017-10-12T14:25:36 INFO src/common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2017-10-12T14:25:36 INFO src/flash_loader.c: Successfully loaded flash loader in sram
 19/19 pages written
2017-10-12T14:25:37 INFO src/common.c: Starting verification of write complete
2017-10-12T14:25:37 INFO src/common.c: Flash written and verified! jolly good!
```

Where is documentation for this?

Nightwalker-87, 2020-03-20:
@MarekLew: What is the current state of this?

Nightwalker-87, 2020-03-31:
A reference to the `--reset` parameter for` st-flash` is obviously missing in the documentation...

commit e250773b (2020-06-17): [doc] Added --reset option for st-flash

(Closes #642)
closed 2020-06-17 by Nightwalker-87