stlink-org/stlink #644: How to read raw flash?

TediumRemedy, 2017-10-14:
Is there a way to read raw flash bytes, for example, if I am saving some config data to the last page of the flash, and later would like to examine it how it changed? Something similar to what ST's STLink utility for Windows does - showing whole flash in hex.
Thanks

xor-gate, 2017-10-24:
You need st-flash tool which does read/writing of STM32 flash.

```
stlinkv2 command line: ./st-flash [--debug] [--reset] [--serial <serial>] [--format <format>] [--flash=<fsize>] {read|write} <path> <addr> <size>
```

Probably do something like (untested):

```
st-flash --format binary read mydump.bin 0x08000000 0x100
```

Nightwalker-87, 2020-02-21:
Feedback closes #644.

closed 2020-02-21 by Nightwalker-87