stlink-org/stlink #646: build and run without root access

ponyatov, 2017-10-18:
I'm trying to build development framework can be used at user level without root access.

Now I have problems with stlink build: 

1. default build wants /usr/local/ but I'm going to avoid any system-wide changes

```
CWD = $(CURDIR)        ## /home/user/STM32
TOOL = $(CWD)/tools    ## cross toolchain and utils will be here
STLINK_CFG = -DCMAKE_INSTALL_PREFIX=/stlink
$(STLINK): stlink/README_ponyatov.md /usr/include/libusb-1.0/libusb.h
	cd stlink ; $(MAKE) clean ;\
	$(MAKE) DESTDIR=$(TMP) CMAKEFLAGS="$(STLINK_CFG)" install &&\
	cp -r $(TMP)/stlink/* $(TOOL)/ 
```
2. Running `$(TOOL)/bin/st-util` gives me error with unfound .so libraries, so I want to build all statically,

Or maybe it will be better just redefine `LD_LIBRARY_PATH` ? I'm going to build fresh `arm-none-eabi-binutils|gcc` from scratch, is it bad to build it statically too ?

Vascom, 2017-10-18:
All static builds are bad. 
Usually root not need for build, only for install. You can just skip install step and run stlink from build directory.

ponyatov, 2017-10-18:
Install works as I need (with some Linux limitation: unprivileged user can not control udev rules).
Also, I found a solution: define `LD_LIBRARY_PATH` and `PATH` in a user profile.

Whats wrong with static build?

xor-gate, 2017-10-18:
We had before a static build (and probably it still is for windows), there is nothing wrong with that. But somebody needed stlink as a shared library.

Nightwalker-87, 2020-03-21:
@ponyatov: What is the current state of this issue? Which release version did you use? Can you please provide more details on your build system.

ponyatov, 2020-03-21:
I have no experience with this a lot of time ago, sorry.

closed 2020-03-21 by ponyatov