stlink-org/stlink #647: flashing a sparse hex file - st-flash erasing untouched block too

linklocal, 2017-10-23:
- [x] Programmer/board type: e.g Stlink/v2
- [x] Programmer firmware version: e.g STSW-LINK007 2.27.56
- [x] Operating system: e.g Linux
- [x] Stlink tools version 6f24bde1aefde8f543861cab8c205e564cb7fb36 v1.3.1
- [x] Stlink commandline tool name: `st-flash`
- [x] Target chip (and optional board): e.g STM32F412CG

memory layout: a 16k bootloader (0x08000000), 16k free (where some persisted data are located) (0x08004000) and the application (starting: 0x08008000) behind that area 
we combined the bootloader (0x08000000-0x08003fff) and application (starting: 0x08008000) to one intel-hex formatted file and flashed it. 
using the st-link (v:4.0.0.0) for windows this works well. the part holding persisted data is untouched. 

using st-link: v1.3.1 on linux side flashing the same file - we see the erasing of the 16k block: 0x08004000-0x08007fff

CMD-LINE:
st-flash --serial 573f70067265575749141467 --format ihex --reset write device_bootloader_firmware.hex

Output:

```
$ st-flash --serial ${ST_LINK_SERIAL} --format=ihex --reset write BUILD/${MCU}/${TOOLCHAIN}/n4m-${ARTICLE}-*-${BOOTLOADER}.hex
2017-10-23T08:30:24 INFO src/common.c: Loading device parameters....
2017-10-23T08:30:24 INFO src/common.c: Device connected is: F4 device, id 0x30006441
2017-10-23T08:30:24 INFO src/common.c: SRAM size: 0x40000 bytes (256 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 16384 bytes
2017-10-23T08:30:24 INFO src/common.c: Attempting to write 431824 (0x696d0) bytes to stm32 address: 134217728 (0x8000000)
EraseFlash - Sector:0x0 Size:0x4000 st-flash 1.3.1-dirty

Flash page at addr: 0x08000000 erasedEraseFlash - Sector:0x1 Size:0x4000 
Flash page at addr: 0x08004000 erasedEraseFlash - Sector:0x2 Size:0x4000 
Flash page at addr: 0x08008000 erasedEraseFlash - Sector:0x3 Size:0x4000 
Flash page at addr: 0x0800c000 erasedEraseFlash - Sector:0x4 Size:0x4000 
Flash page at addr: 0x08010000 erasedEraseFlash - Sector:0x4 Size:0x4000 
Flash page at addr: 0x08014000 erasedEraseFlash - Sector:0x4 Size:0x4000 
Flash page at addr: 0x08018000 erasedEraseFlash - Sector:0x4 Size:0x4000 
Flash page at addr: 0x0801c000 erasedEraseFlash - Sector:0x5 Size:0x4000 
Flash page at addr: 0x08020000 erasedEraseFlash - Sector:0x5 Size:0x4000 
...
```

Expected/description:
we expect that the hexfile is been analysed in content and only blockes containing data are erased and flashed (like the windows version)

Thank you,
carsten

xor-gate, 2017-10-24:
You are the first one which reports using of sparse hex files. It is possible it doesn't work as it is not official supported.

linklocal, 2017-11-06:
thank you. solving that issue would be really helpful for us 
 carsten :)

Nightwalker-87, 2021-12-04:
Closed due to inactivity and missing interest.

closed 2021-12-04 by Nightwalker-87