stlink-org/stlink #649: Solve a problem.

I-n-v-i-n-c-i-b-l-e, 2017-10-27:
Today, I use a ST-Link v2 to download to my board ( STM32F103C8 ). The st-flash always report error and can't identify my board. Then, I try to it on Windows ( ST-Link Utility ). The ST-Link Utility report `Cannot read memory! Disable Read Out Protection and retry.` [Discuss]( https://community.st.com/thread/8710 ).

Finally, go to Target > Option Bytes > Read Out Protection and select `Disable` option. Take it back to OS X, it can be work.

I found some similar problem in this issue, which may help you solve the problem.

So, what is the Read Out Protection?

xor-gate, 2017-11-13:
Read out protection is a feature of the microcontroller to prevent 3th party and malicious users reading out your application from flash memory. So they are not able to copy your products firmware.

Nightwalker-87, 2020-02-21:
Feedback closes #649.

closed 2020-02-21 by Nightwalker-87