stlink-org/stlink #650: ERROR src/commin.c: map_file()==-1

renn0xtek9, 2017-10-28:
I got an error during flash operation. I don't know what it is/means.
The compilation has no problem. Flashing another binary (a simple thing toggling a LED) is not problem either.

- [x ] Programmer/board type: Stlink/v2, clone
- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [x ] Operating system: e.g linux ubuntu 16.04
- [ x] Stlink tools version and/or git commit hash: e.g git-a2a707e4f75fd8aafce3fa6095dc34206c5bb801
- [x ] Stlink commandline tool name: e.g `st-info`, `st-flash write STM32USASRT_IR_StdPeriph.bin 0x8000000`
- [x] Target chip (and optional board): e.g STM32F103C8T6

A as-detailed description possible of the problem with debug output when available.

Output:

```
2017-10-28T19:54:26 ERROR src/common.c: map_file() == -1

```

Expected/description:
`short description of the expected value`

Thank you,
The stlink project maintainers

xor-gate, 2017-11-02:
I'm not sure why the mapping of the binary file to memory fails at your system. The line is located here: https://github.com/texane/stlink/blob/a2a707e4f75fd8aafce3fa6095dc34206c5bb801/src/common.c#L1164

ugifractal, 2019-03-15:
not sure if your case same with mine.
but if st-flash integrated with other tools, when filename parameter is blank, above error may raised.

here is mine
```
sugiartos-MacBook-Pro:~ sugiarto$ st-flash write ""  0x8000000
st-flash 1.5.1-18-g522188d
2019-03-15T13:41:09 INFO common.c: Loading device parameters....
2019-03-15T13:41:09 INFO common.c: Device connected is: F1 Medium-density device, id 0x20036410
2019-03-15T13:41:09 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 1024 bytes
open() == -1
2019-03-15T13:41:09 ERROR common.c: map_file() == -1
stlink_fwrite_flash() == -1
sugiartos-MacBook-Pro:~ sugiarto$ 
```

chenguokai, 2020-04-12:
@Nightwalker-87 I am not quite sure about what I need to do.
If the original issue is indeed the one caused by an empty/non-existing filename, I will be able to fix. Otherwise, without further information, I cannot figure out the problem.

Nightwalker-87, 2020-04-12:
Can you look if there are any issues with filenames that can occur here due to a bug? As suggested in #879 one could also improve some error messages to be more human-readable. In the first place we should make sure that there is no bug present. The error should only be thrown if there _really_ is an invalid filename e.g. as in the second case, which I would not consider a bug at all. Apart from that I feel there is nothing more we can do here currently...

chenguokai, 2020-04-12:
I checked the functions about filenames, no bugs found. The PR provides a more unified way of displaying errors for map_file.

closed 2020-04-12 by Nightwalker-87

Nightwalker-87, 2020-04-12:
Error message has been improved for better understanding. Should there be a reproducible problem related to the used device, please feel free to submit a new ticket.