stlink-org/stlink #651: Couldn't find any ST-Link/V2 devices

me21, 2017-10-31:
- [X] Programmer/board type: Stlink-v2
- [X] Programmer firmware version: V2J28S7
- [X] Operating system: Windows 7 SP1 x64
- [X] Stlink tools version and/or git commit hash: v1.3.1
- [X] Stlink commandline tool name: `st-util`

Output:

```
st-util 1.3.1
2017-10-31T12:24:26 WARN src\usb.c: Couldn't find any ST-Link/V2 devices
```

Expected/description:
Should find my ST-Link/V2. The STM32 ST-LINK utility does.

me21, 2017-11-01:
Yes, it still doesn't work.

In addition, here's output of st-info --chip-id:

```
libusb: warning [init_device] could not get node connection information for device '\\.\USB#VID_0483&PID_3748#6&22FF7FA2&0&3': [87] Parameter is incorrect.
libusb: warning [init_device] could not get node connection information for device '\\.\USB#VID_093A&PID_2510#6&22FF7FA2&0&4': [87] Parameter is incorrect.
```

xor-gate, 2017-11-01:
Probably the builtin libusb has some problems on you platform. I'm not sure which libusb version I used in the stlink v1.3.1 release.

me21, 2017-11-01:
Solved by plugging ST-Link adapter into USB 2.0 port. Didn't work in USB 3.0 port.

closed 2017-11-01 by me21

me21, 2017-11-01:
After upgrading USB 3.0 host controller driver, the utility works with both 2.0 and 3.0 USB ports.

xor-gate, 2017-11-02:
@me21 thanks for investigating and reporting back! luckily it works now for your platform.