stlink-org/stlink #653: Cannot find st-util in clone folder

SuperCiocia, 2017-11-13:
Hello,

I have followed your instructions to clone and make the folder but I cannot find the <code>st-util</code> executable, essential for further work.

All the files I see in the cloned directory are:
[image]

The steps that I have taken till here are:

1) <code> git clone https://github.com/texane/stlink.git </code>

2) <code> cd stlink </code>

3) <code> make </code>

4) <code> cd build/Release && make install DESTDIR=_install

Vascom, 2017-11-14:
You can find it in `src/gdbserver/st-util`

Nightwalker-87, 2020-02-19:
No bug. Issue has been solved.

closed 2020-02-19 by Nightwalker-87