stlink-org/stlink #654: flash loader fails on STM32L0xx

MightyPork, 2017-11-16:
This is a resurgence of issue https://github.com/texane/stlink/issues/356, which is closed and locked (why?).

Using Nucleo L073RZ.
It does not happen with any other chips I tried. 

ST_link firmware is V2J29M18

```none
$ st-flash write build/bees.bin 0x8000000
st-flash 1.4.0
2017-11-16T10:09:01 INFO src/common.c: Loading device parameters....
2017-11-16T10:09:01 INFO src/common.c: Device connected is: L0x Category 5 device, id 0x20086447
2017-11-16T10:09:01 INFO src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x30000 bytes (192 KiB) in pages of 128 bytes
2017-11-16T10:09:01 INFO src/common.c: Ignoring 96 bytes of 0x00 at end of file
2017-11-16T10:09:01 INFO src/common.c: Attempting to write 31904 (0x7ca0) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08007c80 erased
2017-11-16T10:09:03 INFO src/common.c: Finished erasing 250 pages of 128 (0x80) bytes
2017-11-16T10:09:03 INFO src/common.c: Starting Half page flash write for STM32L core id
2017-11-16T10:09:03 INFO src/flash_loader.c: Successfully loaded flash loader in sram
2017-11-16T10:09:06 ERROR src/flash_loader.c: flash loader run error
2017-11-16T10:09:06 WARN src/common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
2017-11-16T10:09:06 WARN src/common.c: 
write_half_pages failed == -1
248/249 pages written
2017-11-16T10:09:36 INFO src/common.c: Starting verification of write complete
2017-11-16T10:09:36 ERROR src/common.c: Verification of flash failed at offset: 0
stlink_fwrite_flash() == -1
```

The flashing is extremely slow, but despite the errors shown, after a board reset with the hardware button, the uploaded firmware seems to be working correctly.

MightyPork, 2017-11-16:
Ah, I see there's a couple old isues with this exact error, closed as "will be fixed in 1.2" or so - alas...

szechyjs, 2018-02-01:
I'm having similar problems

```
st-flash 1.4.0
2018-02-01T18:02:36 INFO src/common.c: Loading device parameters....
2018-02-01T18:02:36 INFO src/common.c: Device connected is: L011 device, id 0x10086457
2018-02-01T18:02:36 INFO src/common.c: SRAM size: 0x2000 bytes (8 KiB), Flash: 0x4000 bytes (16 KiB) in pages of 128 bytes
2018-02-01T18:02:36 INFO src/common.c: Attempting to write 12508 (0x30dc) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08003080 erased
2018-02-01T18:02:37 INFO src/common.c: Finished erasing 98 pages of 128 (0x80) bytes
2018-02-01T18:02:37 INFO src/common.c: Starting Half page flash write for STM32L core id
2018-02-01T18:02:37 ERROR src/flash_loader.c: unknown coreid, not sure what flash loader to use, aborting! coreid: bc11477, chipid: 457
2018-02-01T18:02:37 WARN src/flash_loader.c: Failed to write flash loader to sram!
2018-02-01T18:02:37 WARN src/common.c: stlink_flash_loader_init() == -1
2018-02-01T18:02:37 WARN src/common.c:
write_half_pages failed == -1
 96/ 97 pages written
2018-02-01T18:02:48 INFO src/common.c: Starting verification of write complete
2018-02-01T18:02:48 INFO src/common.c: Flash written and verified! jolly good!
```

commit 4e270606 (2018-02-28): Fix missing flash_loader for L011 (#675)

Fixes #654 flash loader on L011
closed 2020-03-18 by Nightwalker-87