stlink-org/stlink #655: Debugging not possible on STM32F407 Discovery board

nedsana, 2017-11-27:
Hi, 

For the last few days I am trying to use the st-util and arm-none-eabi-gdb with my code, but without success. The problem is that gdb will never stop on breakpoints I am setting. To be more precise, the breakpoints seem to be hit and the execution of the MCU is interrupted (I have an interrupt driven LED blinking which is stopped). Unfortunately GDB will not understand this and will not return the control to me. 

I have tried this on Ubuntu and Linux Mint. 

I am using the common STM32F407 Discovery with on-board ST_Link chip (ST-LINK/V2).

I can load the and erase the chip. 

The version of st-util is st-util 1.4.0-12-g95b6e03-dirty (master branch).  I tried to use an older version 1.3.0, but I can not compile this one - I get "undefined reference" linker error for libusb_close, libusb_exit, libusb_init, etc. And this is weird as version 1.4.0 compiles fine.

I have the following .gdbinit script:

> define reload
> kill
> monitor jtag_reset
> monitor halt
> load
> monitor reset halt
> stepi
> end
> 
> target extended localhost:4242
> monitor halt
> load
> monitor reset halt
> stepi

Output st-util:

> lin@lin-virtual-machine ~/Projects/STM32CubeMX/Projects/STM32F407/STM32F407 $ st-util 
> st-util 1.4.0-12-g95b6e03-dirty
> 2017-11-27T10:02:42 INFO common.c: Loading device parameters....
> 2017-11-27T10:02:42 INFO common.c: Device connected is: F4 device, id 0x10016413
> 2017-11-27T10:02:42 INFO common.c: SRAM size: 0x30000 bytes (192 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 16384 bytes
> 2017-11-27T10:02:42 INFO gdb-server.c: Chip ID is 00000413, Core ID is  2ba01477.
> 2017-11-27T10:02:42 INFO gdb-server.c: Listening at *:4242...
> 2017-11-27T10:02:45 INFO gdb-server.c: Found 6 hw breakpoint registers
> 2017-11-27T10:02:45 INFO gdb-server.c: GDB connected.
> 2017-11-27T10:02:46 INFO common.c: Attempting to write 16384 (0x4000) bytes to stm32 address: 134217728 (0x8000000)
> Flash page at addr: 0x08000000 erased
> 2017-11-27T10:02:47 INFO common.c: Finished erasing 1 pages of 16384 (0x4000) bytes
> 2017-11-27T10:02:47 INFO common.c: Starting Flash write for F2/F4/L4
> 2017-11-27T10:02:47 INFO flash_loader.c: Successfully loaded flash loader in sram
> enabling 32-bit flash writes
> size: 16384
> 2017-11-27T10:02:47 INFO common.c: Starting verification of write complete
> 2017-11-27T10:02:47 INFO common.c: Flash written and verified! jolly good!
> 2017-11-27T10:02:47 INFO common.c: Attempting to write 16384 (0x4000) bytes to stm32 address: 134234112 (0x8004000)
> Flash page at addr: 0x08004000 erased
> 2017-11-27T10:02:48 INFO common.c: Finished erasing 1 pages of 16384 (0x4000) bytes
> 2017-11-27T10:02:48 INFO common.c: Starting Flash write for F2/F4/L4
> 2017-11-27T10:02:48 INFO flash_loader.c: Successfully loaded flash loader in sram
> enabling 32-bit flash writes
> size: 16384
> 2017-11-27T10:02:48 INFO common.c: Starting verification of write complete
> 2017-11-27T10:02:49 INFO common.c: Flash written and verified! jolly good!
> 2017-11-27T10:02:49 INFO gdb-server.c: Found 6 hw breakpoint registers

Output GDB:

> lin@lin-virtual-machine ~/Projects/STM32CubeMX/Projects/STM32F407/STM32F407 $ arm-none-eabi-gdb build/STM32F407.elf
> GNU gdb (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 7.12.1.20170417-git
> Copyright (C) 2017 Free Software Foundation, Inc.
> License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
> This is free software: you are free to change and redistribute it.
> There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
> and "show warranty" for details.
> This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
> Type "show configuration" for configuration details.
> For bug reporting instructions, please see:
> <http://www.gnu.org/software/gdb/bugs/>.
> Find the GDB manual and other documentation resources online at:
> <http://www.gnu.org/software/gdb/documentation/>.
> For help, type "help".
> Type "apropos word" to search for commands related to "word"...
> Reading symbols from build/STM32F407.elf...done.
> Reset_Handler () at startup_stm32f407xx.s:78
> 78	  ldr   sp, =_estack     /* set stack pointer */
> Loading section .isr_vector, size 0x188 lma 0x8000000
> Loading section .text, size 0x7220 lma 0x8000188
> Loading section .rodata, size 0x8c lma 0x80073a8
> Loading section .ARM, size 0x8 lma 0x8007434
> Loading section .init_array, size 0x4 lma 0x800743c
> Loading section .fini_array, size 0x4 lma 0x8007440
> Loading section .data, size 0x1bc lma 0x8007444
> Start address 0x8007138, load size 30208
> Transfer rate: 10 KB/sec, 3776 bytes/write.
> Reset_Handler () at startup_stm32f407xx.s:81
> 81	  movs  r1, #0
> (gdb) break main.c:159
> Breakpoint 1 at 0x8002006: file Src/main.c, line 159.
> (gdb) continue
> Continuing.
> Note: automatically using hardware breakpoints for read-only addresses.

I you have any suggestions please let me know. I am really stuck here.

Thank you in advance, 
Greetings, Nedelin

xor-gate, 2017-11-27:
It is possible it will not return from the interrupt and you are the first one reporting this. Could you try with [openocd](https://github.com/ntfreak/openocd), it is more widely used and more complete than texane/stlink?

nedsana, 2017-11-27:
OK, I will try with OpenOCD and will let you know. In the meantime if you need my setup to reproduce the issue, please let me know.
Thank you.

nedsana, 2017-11-29:
Unfortunately OpenOCD did not solve my problem. The behavior seems the same. Do you have any suggestions? 
Is it possible that my board is broken? 
Or maybe my code is causing this issues? But I have tried to use breakpoints in simple blinking example and they are also not hit.

xor-gate, 2017-11-30:
That's very weird, I can't say directly what causes the problem or where you have to look at. OpenOCD is a fairly stable piece of software and if it doesn't work something else is wrong. You must make sure you compile your code with debug symbols `-ggdb` flag and without optimalisation `-O0` (zero).

Nightwalker-87, 2020-03-20:
@nedsana: Please verify if the problem still exists in Release v1.5.0.

nedsana, 2020-03-20:
Hi,

I am not working this project any longer (for more then 2.5 years) and my
setup is long gone.
So I can not make any verification. I am sorry for this. I guess you can
consider this issue resolved and close it.

Greetings, Nedelin.

Nightwalker-87, 2020-03-20:
Thx for your feedback. As one of the new contributors of this project I am currently reviewing leftovers from the recent years. Sorry to see that this ticket seems to have gone down the sink without any further feedback to you. Have you found any solution or workaround however you can still remember?

nedsana, 2020-03-20:
Hi,
I think I never solved the problem. Besides we focused on using esp32
device, but a few months later the whole project was put aside ...
Greetings, Nedelin

Nightwalker-87, 2020-07-18:
Can someone reproduce this step by step and report back?
I propose to use the `v1.5.0-patch` branch for this.

Nightwalker-87, 2021-04-04:
Having reviewed changes related to st-util and the `gdb-server`, this issue still appears to be unresolved as of v1.6.1.
Some relevant changes may be found in commits 34c0a31204d33fd15a1186709e3f60d186bcda3e and 09ea99aea51f741ce9c8c0e2ad755d40b3d86021.

Thus this should be investigated further on the current `develop` branch.

Nightwalker-87, 2021-04-08:
@Ant-ON I can't locate any suspicious behaviour here. Have I missed something when looking at this?

Ant-ON, 2021-04-08:
@Nightwalker-87 This is a tricky problem. The target program did not run in the debugger. And in throwgh OpenOСD too. Perhaps there were some features of the software or hardware that did not allow this to be done. It is hard to say.

Nightwalker-87, 2021-04-11:
I expect this issue still to be present in v1.5.0 ans v1.5.1.

Nightwalker-87, 2021-05-13:
Can someone with the same or similar hardware please check and try to reproduce this issue with the older versions mentioned above?

Nightwalker-87, 2021-05-17:
Drawing this issue to a close, we shall put on record, that it indeed seems quiet unlikely for two independent and different toolsets to cause the same or at least a very similar problem. Instead the issue suggests the observed behaviour to be caused by the `gdb` tool itself. If still relevant we can thus give the advice to refer to the respective project for further support on this topic.

closed 2021-05-17 by Nightwalker-87