stlink-org/stlink #656: Flash upload through GPIO

feliwir, 2017-11-27:
Feature Request

- [X ] Operating system: Linux

Hello everyone,

as the title says I'd like to upload the compiled program via GPIO pins to my board/STM chip. I have used ATmega chips before and the program "avrdude" allowed this by connecting the GPIO pins to the according SPI pins (see [here](https://learn.adafruit.com/program-an-avr-or-arduino-using-raspberry-pi-gpio-pins/overview) ). This would be useful for me so i can upload the program with a Raspberry Pi for example. 
I don't know how much effort it is to use the LinuxGPIO system and if STM32 chips are programable like this, so I'd be happy to get an opinion on this.
Thanks

xor-gate, 2017-11-27:
ARM cortex only allows uploading through JTAG pins/Serial-wire-debug pins (SWCLK/SWDIO). A stlink programmer only costs approximate 15 euro and can `st-flash` can be run on the raspberry pi. Or you should get a nucleo/discovery board which is a development kit and can hook up the programmer to another board.

closed 2017-11-27 by xor-gate

feliwir, 2017-11-27:
@xor-gare So why is it not possible to write to those pins directly via GPIO?

xor-gate, 2017-11-28:
As the name suggests GPIO is General Purpose Input/Output and the STM32 has a JTAG/SWD peripherial. You are able to bitbang this protocols with GPIO, but that is not the purpose of the `texane/stlink` project which only support STLink programmers which talk the SWD protocol (and not the JTAG protocol). GPIO bitbanging protocols can introduce timing problems from user-space (e.g linux). That is why dedicated peripherials are used which make sure the bit/clock timings are correct.

Hopefully you understand now a little bit more.