stlink-org/stlink #658: st-util gdb script extremely slow on Linux?

mbriday, 2017-12-07:
Hi,

- [X] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
* On-board ST-LINK/V2-1 => Nucleo F303

```
st-util 1.4.0
2017-12-07T15:09:13 INFO src/common.c: Loading device parameters....
2017-12-07T15:09:13 INFO src/common.c: Device connected is: F334 device, id 0x10016438
2017-12-07T15:09:13 INFO src/common.c: SRAM size: 0x3000 bytes (12 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 2048 bytes
2017-12-07T15:09:13 INFO src/gdbserver/gdb-server.c: Chip ID is 00000438, Core ID is  2ba01477.
```

- [X] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
* Linux (Ubuntu 16.04 LTS)
* Mac OS X (Yosemite).

- [x] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
* version 1.4.0 on Linux-Ubuntu 16.04
* version 1.3.1-19-g55c0572 on Mac OS X (Yosemite)

- [X] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util`
=> st-util

- [X] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)
[STM32F303 Nucleo-32](http://www.st.com/en/evaluation-tools/nucleo-f303k8.html)

I am using st-util to send a GDB script to the STM32F303, both on Mac and Linux. The script executes only very few instructions on the target but read/write registers (there is a script examble below) and most of the time is done to process gdb commands.
On the Mac (with version 1.3.1), it takes about 30s to execute the whole script… but more than 18min30s on the Linux laptop (which is more recente with more RAM).
The simple example below takes nearly 3.5s on the Linux, and more than ten times less on the Mac.

I compiled the Linux from the sources, and use the libusb provided by Ubuntu.

Is there a reason why the Linux host works so slowly?
regards,
Mik

PS: The [code provided](https://github.com/texane/stlink/files/1539455/code.zip) should not be executed directly, but with the script.

```
tar extended-remote :4242
set interactive-mode off
load
define dumpRegs
	printf "0x%08x\t",$r0
	printf "0x%08x\t",$r1
	printf "0x%08x\t",$r2
	printf "0x%08x\t",$r3
	printf "0x%08x\t",$r4
	printf "0x%08x\t",$r5
	printf "0x%08x\t",$r6
	printf "0x%08x\t",$r7
	printf "0x%08x\t",$r8
	printf "0x%08x\t",$r9
	printf "0x%08x\t",$r10
	printf "0x%08x\t",$r11
	printf "0x%08x\t",$r12
	printf "0x%08x\t",$sp
	printf "0x%08x\t",$lr
	printf "0x%08x\t",$pc
	printf "0x%08x\t",$cpsr
end

set $sp=0x8
set $lr=0xdead
set logging file tst.json_output.gdb
set logging redirect on
set logging on

set $oldpc = $pc
set $r0=0
set $r0=0
set $cpsr=1090519040
dumpRegs()
printf "A\n"
set logging off
si
set logging on
dumpRegs()
printf "B\n"

set $pc = $oldpc
set $r0=305419896
set $r0=0
set $cpsr=1090519040
dumpRegs()
printf "A\n"
set logging off
si
set logging on
dumpRegs()
printf "B\n"

set $pc = $oldpc
set $r0=4294967295
set $r0=0
set $cpsr=1090519040
dumpRegs()
printf "A\n"
set logging off
si
set logging on
dumpRegs()
printf "B\n"

set logging off
quit
```

Nightwalker-87, 2020-03-21:
@mbriday: Can you re-test and verify your observation with a clean build of Release v1.4.0? Should there be no change I'd suggest to retry with v1.5.0 and provide feedback to us.

Nightwalker-87, 2020-07-05:
Can someone run a similar test script with `st-util` and `gdb` on our `v1.3.1-patch`, `v1.4.0-patch` and `1.5.0-patch` branches in order to compare? It would be helpful to know, if we were facing a regression here. Also one should be able to rule out any influence on even later versions.

Nightwalker-87, 2021-04-04:
This could either have been related to different `libusb` versions used on both machines, or different toolchains (clang, gcc) for building the latter, or even related to specific local hardware issues, that did not have any impact on the correct command and program execution itself.
As we have no chance to reproduce or verify this and together with the knowledge that there were no general performance issues on linux-based systems in the past, this issue can now be safely closed as resolved.

closed 2021-04-04 by Nightwalker-87