stlink-org/stlink #659: Wrong flash size determined after debugging

martonmiklos, 2017-12-11:
- [X] Programmer/board type: Stlink/v2
- [X] Programmer firmware version: e.g STSW-LINK007 2.29.18 
- [X ] Operating system: Linux: Ubuntu 14.04 
- [X ] Stlink tools version and/or git commit hash: 19691485359afef1a256964afcbb8dcf4b733209
- [X ] Stlink commandline tool name: st-flash
- [X ] Target chip: STM32F103C8T8

After performing a debug session with openOCD+GDB I kill both process (OpenSTM32 does that).
Afterwards when I run the st-flash I got malformed readout and wrong flash size:

Output:

```
mm@mm-imac:~/Projektek/stlink$ ./st-flash --flash=64k read /tmp/out.hex 0x8004000 0xC000
st-flash 1.4.0-13-g1969148
2017-12-11T17:08:50 INFO common.c: Loading device parameters....
2017-12-11T17:08:50 INFO common.c: Device connected is: F1 Medium-density device, id 0x20036410
2017-12-11T17:08:50 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: **0x1904000** bytes **(25616 KiB)** in pages of 1024 bytes
Forcing flash size: --flash=0x00010000
mm@mm-imac:~/Projektek/stlink$ ./st-flash --flash=64k read /tmp/out.hex 0x8004000 0xC000
st-flash 1.4.0-13-g1969148
2017-12-11T17:08:54 INFO common.c: Loading device parameters....
2017-12-11T17:08:54 INFO common.c: Device connected is: F1 Medium-density device, id 0x20036410
2017-12-11T17:08:54 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
```
If I run the st-flash again it reads the correct flash size. 

Did anyone faced with  the similar issue?

Nightwalker-87, 2020-03-21:
@martonmiklos: Any progress on this?

Nightwalker-87, 2020-04-14:
Possibly related to issue 275.

chenguokai, 2020-04-14:
Some clues that I have found: 
0x6410 is one magic number used in openocd.
The decimal form, 25616, can be found as a part of the name of a SRAM chip, IS61WV25616EDBLL.

The one used in openocd is quite suspicious.

Nightwalker-87, 2020-06-24:
> Please verify if the problem still exists in Release v1.6.0.

@martonmiklos: Have you tested this? Please let us know.

martonmiklos, 2020-06-30:
I do not use the openOCD anymore, and I no longer have that STM32F103C8T8 based gear where this issue happened. I am closing this issue, in the case someone ran into this could be reopened and investigated again.

Nightwalker-87, 2020-06-30:
@martonmiklos: Please don't close tickets and leave this task to project maintenance.
This may prevent others from dealing with this and interferes with general ticket handling.
It's ok though, if you don't want to be involved in this issue any more.
We can take that into account and you can unsubscribe from this issue to avoid be notified on updates any longer.

swiftgeek, 2020-07-11:
Same issue but only with very particular unit of the STM32F103C8T8

Nightwalker-87, 2020-07-17:
@swiftgeek: Thanks for the feedback. I think this is not a satisfiable state and thus should be investigated.

Nightwalker-87, 2020-12-14:
@chenguokai: I still don't really understand what you explained before. What does that mean?

Nightwalker-87, 2021-03-12:
I followed the steps of @martonmiklos with the difference that I performed the debug session via `gdb-multiarch` in combination with `st-util` as described in our tutorial. The result below reveals that I was not able to reproduce the issue with v1.5.1:

```
$ sudo st-flash --flash=64k read /tmp/out.hex 0x8004000 0xC000
st-flash 1.5.1-1-gffa986c-dirty
2021-03-12T19:14:27 INFO common.c: Loading device parameters....
2021-03-12T19:14:27 INFO common.c: Device connected is: F1 Medium-density device, id 0x20036410
2021-03-12T19:14:27 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
$ sudo st-flash --flash=64k read /tmp/out.hex 0x8004000 0xC000
st-flash 1.5.1-1-gffa986c-dirty
2021-03-12T19:14:31 INFO common.c: Loading device parameters....
2021-03-12T19:14:31 INFO common.c: Device connected is: F1 Medium-density device, id 0x20036410
2021-03-12T19:14:31 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
```

I had to use v1.5.1 as I was not able to compile versions v1.4.0 and v1.5.0.
Nevertheless I may have a look if anything relevant to this topic happened between 19691485359afef1a256964afcbb8dcf4b733209 and v1.5.1.

Nightwalker-87, 2021-03-26:
I couldn't find anything relevant from the git tree between the mentioned commits.
The reporting of wrong flash sizes may appear on single chips which are partly defective or faked.
As this issue could not be reproduced we can now safely close it as resolved.

closed 2021-03-26 by Nightwalker-87