stlink-org/stlink #660: unknown chip id! 0x30003 , STM32F469 Please help

nquantum, 2017-12-19:
I use stlink in ubuntu 16.04 LTS and compile stlink from source: 
 - git clone https://github.com/texane/stlink.git
 - cd stlink
 - make release
 - make debug

Then I flash my STM32F469NI Discovery Board by:
  ~/tmp/stlink/build/Release$ ./st-flash write ~/ros_uart.hex 0x8000000

This is output: 
  st-flash 1.4.0-13-g1969148
  2017-12-19T12:22:45 INFO common.c: Loading device parameters....
  2017-12-19T12:22:45 WARN common.c: unknown chip id! 0x30003

I though stm32f469 already supported?? Since I search in issues here [https://github.com/texane/stlink/pull/345](url) But I seem to be not yet? 
How can I do? Please advise.

nquantum, 2017-12-19:
After more search I found that STM32F469 already supported and after I tracked the code to confirm that.
Some where in issues said that unknown chip id! can try with erase first then write again , it work.
Also I should not :
./st-flash write ~/ros_uart.hex 0x8000000
But I should from [https://github.com/texane/stlink/blob/master/doc/tutorial.md](url) :
./st-flash --format ihex write ~/ros_uart.hex

Now work great!

closed 2017-12-19 by nquantum

xor-gate, 2017-12-19:
Good to hear it works now!