stlink-org/stlink #661: Flashing STM32F0 fails on first flashing instance

ksschwabe, 2018-01-10:
- [x ] Programmer: Stlink/v2
- [ x] Operating system: Ubuntu 16.04
- [ x] Stlink tools git commit hash: 19691485359afef1a256964afcbb8dcf4b733209
- [ x] Stlink commandline tool name: `st-flash`
- [x ] Target chip: STM32F051R8T6TR and STM32F030K6T6

Whenever I flash any of the F0 chips that I have, the flashing fails on the first execution. I have to reset the chip (power off and power back on again) and try to flash the chip again. On the second execution it successfully flashes the chip.

Below, I have included the standard console output and the debug output for both the initial failure and the subsequent successful flashing. This seems to be related to issue #92, but it doesn't look as if simply running st-flash directly works on the fist execution.

Initial flashing:
```
k@k-Desktop:~/FW$  st-flash write FUPS.bin 0x08000000
st-flash 1.4.0-13-g1969148
2018-01-10T11:44:05 INFO common.c: Loading device parameters....
2018-01-10T11:44:05 INFO common.c: Device connected is: F0 device, id 0x20006440
2018-01-10T11:44:05 INFO common.c: SRAM size: 0x2000 bytes (8 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2018-01-10T11:44:05 INFO common.c: Attempting to write 35648 (0x8b40) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08008800 erased
2018-01-10T11:44:06 INFO common.c: Finished erasing 35 pages of 1024 (0x400) bytes
2018-01-10T11:44:06 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2018-01-10T11:44:06 INFO flash_loader.c: Successfully loaded flash loader in sram
 35/35 pages written
2018-01-10T11:44:08 INFO common.c: Starting verification of write complete
2018-01-10T11:44:08 ERROR common.c: Verification of flash failed at offset: 0
stlink_fwrite_flash() == -1
```
After having reset the chip (power off and power back on):
```
k@k-Desktop:~/FW$  st-flash write FUPS.bin 0x08000000
st-flash 1.4.0-13-g1969148
2018-01-10T11:44:14 INFO common.c: Loading device parameters....
2018-01-10T11:44:14 INFO common.c: Device connected is: F0 device, id 0x20006440
2018-01-10T11:44:14 INFO common.c: SRAM size: 0x2000 bytes (8 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2018-01-10T11:44:14 INFO common.c: Attempting to write 35648 (0x8b40) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08008800 erased
2018-01-10T11:44:15 INFO common.c: Finished erasing 35 pages of 1024 (0x400) bytes
2018-01-10T11:44:15 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2018-01-10T11:44:15 INFO flash_loader.c: Successfully loaded flash loader in sram
 35/35 pages written
2018-01-10T11:44:16 INFO common.c: Starting verification of write complete
2018-01-10T11:44:17 INFO common.c: Flash written and verified! jolly good!
```

aric1987, 2018-01-24:
more info:
st-flash --debug --reset write .encrypt.bin 0x08000000
st-flash 1.4.0
2018-01-24 DEBUG src/common.c: stlink current mode: mass
2018-01-24 DEBUG src/common.c: stlink current mode: mass
2018-01-24 DEBUG src/common.c: *** stlink_enter_swd_mode ***
2018-01-24 DEBUG src/common.c: *** looking up stlink version
2018-01-24 DEBUG src/common.c: st vid         = 0x0483 (expect 0x0483)
2018-01-24 DEBUG src/common.c: stlink pid     = 0x3748
2018-01-24 DEBUG src/common.c: stlink version = 0x2
2018-01-24 DEBUG src/common.c: jtag version   = 0x11
2018-01-24 DEBUG src/common.c: swim version   = 0x4
2018-01-24 DEBUG src/common.c: *** stlink_jtag_reset ***
2018-01-24 DEBUG src/common.c: *** stlink_reset ***
2018-01-24 INFO src/common.c: Loading device parameters....
2018-01-24 DEBUG src/common.c: *** stlink_core_id ***
2018-01-24 DEBUG src/common.c: core_id = 0x0bb11477
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 0 is 0xe0042000
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 10006444 is 0x40015800
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 ffff0020 is 0x1ffff7cc
2018-01-24 INFO src/common.c: Device connected is: F0 small device, id 0x10006444
2018-01-24 INFO src/common.c: SRAM size: 0x1000 bytes (4 KiB), Flash: 0x8000 bytes (32 KiB) in pages of 1024 bytes
2018-01-24 DEBUG src/common.c: *** set_swdclk ***
2018-01-24 DEBUG src/common.c: stlink current mode: debug (jtag or swd)
2018-01-24 DEBUG src/common.c: stlink current mode: debug (jtag or swd)
2018-01-24 DEBUG src/common.c: *** stlink_jtag_reset ***
2018-01-24 DEBUG src/common.c: *** stlink_reset ***
2018-01-24 DEBUG src/common.c: *** stlink_force_debug_mode ***
2018-01-24 DEBUG src/common.c: *** stlink_status ***
2018-01-24 DEBUG src/common.c:   core status: halted
2018-01-24 INFO src/common.c: Ignoring 4 bytes of 0xff at end of file
2018-01-24 INFO src/common.c: Attempting to write 23136 (0x5a60) bytes to stm32 address: 134217728 (0x8000000)
2018-01-24 DEBUG src/common.c: *** stlink_core_id ***
data_len = 2 0x2
 77 14

2018-01-24 DEBUG src/common.c: core_id = 0x0bb11477
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x4002200c
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 80 is 0x40022010
2018-01-24 DEBUG src/common.c: *** stlink_write_debug32 45670123 to 0x40022004
2018-01-24 DEBUG src/common.c: *** stlink_write_debug32 cdef89ab to 0x40022004
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x40022010
2018-01-24 DEBUG src/common.c: Successfully unlocked flash
2018-01-24 DEBUG src/common.c: *** stlink_write_debug32 2 to 0x40022010
2018-01-24 DEBUG src/common.c: *** stlink_write_debug32 8000000 to 0x40022014
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 2 is 0x40022010
2018-01-24 DEBUG src/common.c: *** stlink_write_debug32 42 to 0x40022010
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 3 is 0x4002200c
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 3 is 0x4002200c
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 3 is 0x4002200c
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 3 is 0x4002200c
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 3 is 0x4002200c
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 3 is 0x4002200c
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 23 is 0x4002200c
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 23 is 0x4002200c
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 20 is 0x4002200c
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 2 is 0x40022010
2018-01-24 DEBUG src/common.c: *** stlink_write_debug32 82 to 0x40022010
Flash page at addr: 0x08005800 erased
2018-01-24 INFO src/common.c: Finished erasing 23 pages of 1024 (0x400) bytes
2018-01-24 INFO src/common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2018-01-24 DEBUG src/common.c: *** stlink_write_mem32 52 bytes to 0x20000000
2018-01-24 INFO src/flash_loader.c: Successfully loaded flash loader in sram
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 82 is 0x40022010
2018-01-24 DEBUG src/common.c: *** stlink_write_debug32 45670123 to 0x40022004
2018-01-24 DEBUG src/common.c: *** stlink_write_debug32 cdef89ab to 0x40022004
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 2 is 0x40022010
2018-01-24 DEBUG src/common.c: Successfully unlocked flash
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 2 is 0x40022010
2018-01-24 DEBUG src/common.c: *** stlink_write_debug32 1 to 0x40022010
[... 41 lines cut ...]
2018-01-24 DEBUG src/common.c: *** stlink_status ***
2018-01-24 DEBUG src/common.c:   core status: running
2018-01-24 DEBUG src/common.c: *** stlink_status ***
2018-01-24 DEBUG src/common.c:   core status: running
2018-01-24 DEBUG src/common.c: *** stlink_status ***
2018-01-24 DEBUG src/common.c:   core status: running
2018-01-24 ERROR src/flash_loader.c: flash loader run error
2018-01-24 ERROR src/common.c: stlink_flash_loader_run(0x8000000) failed! == -1
2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 20001000 is 0x8000000
2018-01-24 DEBUG src/common.c: *** stlink_write_reg
data_len = 2 0x2
 81 00

2018-01-24 DEBUG src/common.c: *** stlink_read_debug32 8005641 is 0x8000004
2018-01-24 DEBUG src/common.c: *** stlink_write_reg
data_len = 2 0x2
 81 00

2018-01-24 DEBUG src/common.c: *** stlink_run ***
stlink_fwrite_flash() == -1
2018-01-24 DEBUG src/common.c: *** stlink_exit_debug_mode ***
2018-01-24 DEBUG src/common.c: *** stlink_write_debug32 a05f0000 to 0xe000edf0
2018-01-24 DEBUG src/common.c: *** stlink_close ***
Makefile:62: recipe for target 'swd_write' failed
make: *** [swd_write] Error 255

Nightwalker-87, 2020-03-20:
@ksschwabe @aric1987: Please verify if the problem still exists in Release v1.5.0.

ksschwabe, 2020-03-31:
This problem now seems to have been resolved in v1.6.0.

Nightwalker-87, 2020-03-31:
Thx for the feedback. Have you also tested v1.5.0 and v1.5.1?
That would help to find the related commit in order to correct the changelog.

Nightwalker-87, 2020-07-05:
This should be tested with our `v1.5.1-patch` branch. Please report if the problem persists.

PR #1113 (2021-03-25): Flash loader rework
In this PR, the waiting for data writing is changed from cycles to ms. This should be a more robust solution. 
Fixed a flash loader bug. The flash loader had an incorrect busy wait cycle. The use of bootloader argument registers during boot removed. Removed magic with recalculation of the `size` to the `count`. The number of bytes to load (`size`) is transferred to the flash loader now.
Added disabling of interrupts during flash loader execution. In some cases interrupts caused a failure

(Closes #356, Closes #556, Closes #593, Closes #597, Closes #607, Closes #612, Closes  [... cut]
closed 2021-03-25 by Nightwalker-87