stlink-org/stlink #662: stlink on linux : ERROR common.c: addr too high

VARSHNED, 2018-01-30:
I am  trying to flash the demo software of STM32F769 discovery kit. But below is the error that i am obtaining

st-flash 1.4.0-13-g1969148
2018-01-30T17:40:14 INFO common.c: Loading device parameters....
2018-01-30T17:40:14 INFO common.c: Device connected is: F76xxx device, id 0x10016451
2018-01-30T17:40:14 INFO common.c: SRAM size: 0x80000 bytes (512 KiB), Flash: 0x200000 bytes (2048 KiB) in pages of 2048 bytes
2018-01-30T17:40:14 INFO common.c: Attempting to write 26556392 (0x19537e8) bytes to stm32 address: 134217728 (0x8000000)
2018-01-30T17:40:14 ERROR common.c: addr too high
stlink_fwrite_flash() == -1

DenisBiz-cloud, 2020-02-06:
Has this problem been solved?

Nightwalker-87, 2020-03-21:
@VARSHNED: Any progress on this? Can you reproduce? If yes, please try Release v1.5.0.

Nightwalker-87, 2020-04-03:
Resolved in #714.

closed 2020-04-03 by Nightwalker-87