stlink-org/stlink #663: STM32F4DISCOVERY: st-flash failure "addr not a multiple of pagesize, not supported"

zambs76, 2018-02-02:
- [ ] Programmer/board type: Stlink/v2
- [ ] Programmer firmware version:  not available/ not known
- [ ] Operating system: Linux Raspbian (on Raspberry Pi 3) Linux raspberry-pi 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux

- [ ] Stlink tools version and/or git commit hash: commit 19691485359afef1a256964afcbb8dcf4b733209 / st-flash 1.4.0-13-g1969148
- [ ] Stlink commandline tool name: `st-flash`
- [ ] Target chip (and optional board): STM32F4DISCOVERY

Hello,
I compiled on Raspberry pi 3 the source of contained in this repository.
Now I am trying to flash an .hex image on the  STM32F4DISCOVERY board

Output:
```

root@raspberrypi:~/stlink# st-info --probe
Found 1 stlink programmers
 serial: 493f70066775545525150167
openocd: "\x49\x3f\x70\x06\x67\x75\x54\x55\x25\x15\x01\x67"
  flash: 1048576 (pagesize: 16384)
   sram: 196608
 chipid: 0x0413
  descr: F4 device

Flashing ...

root@raspberrypi:~/stlink# st-flash --format ihex write application.hex
st-flash 1.4.0-13-g1969148
2018-02-02T12:38:01 INFO common.c: Loading device parameters....
2018-02-02T12:38:01 INFO common.c: Device connected is: F4 device, id 0x10076413
2018-02-02T12:38:01 INFO common.c: SRAM size: 0x30000 bytes (192 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 16384 bytes
2018-02-02T12:38:01 INFO common.c: Attempting to write 21460 (0x53d4) bytes to stm32 address: 134529024 (0x804c000)
2018-02-02T12:38:01 ERROR common.c: addr not a multiple of pagesize, not supported
stlink_fwrite_flash() == -1
```

Expected/description:
Flash should complete with string "Flash written and verified! jolly good!" as we are trying to write to an address  (0x804c000) which is multiple of the page size (16384/ 4000Hex)

Thanks a lot!

chenguokai, 2020-04-20:
I can reproduce this problem on my F401 board, with start address set to 0x803c000 (0x804c000 is out of bound in my board).

I am doing some initial testing. The current finding indicates a misconfigured page size. To be clear, the page size info is right when `st-flash` prints 
```
2020-04-20T19:34:20 INFO common.c: SRAM size: 0x10000 bytes (64 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 16384 bytes
```

with src/common.c:

```c
ILOG("SRAM size: %#x bytes (%d KiB), Flash: %#x bytes (%d KiB) in pages of %u bytes\n",
            sl->sram_size, sl->sram_size / 1024, sl->flash_size, sl->flash_size / 1024,
	 (unsigned int)sl->flash_pgsz);
```

The page size changes, however when doing the alignment check.

```c
    } else if (addr & (sl->flash_pgsz - 1)) {
        printf("DEBUG: addr=0x%x, st->flash_pgsz=0x%lx\n", addr, sl->flash_pgsz);
        // DEBUG line is added myself for testing
        ELOG("addr not a multiple of pagesize, not supported\n");
        return -1;
    }
```

This debug output would give 
```
DEBUG: addr=0x803c000, st->flash_pgsz=0x20000
2020-04-20T19:34:20 ERROR common.c: addr not a multiple of pagesize, not supported
```

grevaillot, 2020-04-20:
i'm quite sure that f401 some bigger pages at then end of its flash - i need to check RM, but that flash start address might be in that area.

grevaillot, 2020-04-20:
From RM:

```
Table 5. Flash module organization (STM32F401xB/C and STM32F401xD/E)

Block Name Block base addresses Size 
Sector 0 0x0800 0000 - 0x0800 3FFF 16 Kbytes 
Sector 1 0x0800 4000 - 0x0800 7FFF 16 Kbytes
Sector 2 0x0800 8000 - 0x0800 BFFF 16 Kbytes
Sector 3 0x0800 C000 - 0x0800 FFFF 16 Kbytes
Sector 4 0x0801 0000 - 0x0801 FFFF 64 Kbytes 
Sector 5 0x0802 0000 - 0x0803 FFFF 128 Kbytes
Sector 6 0x0804 0000 - 0x0805 FFFF 128 Kbytes
Sector 7 0x0806 0000 - 0x0807 FFFF 128 Kbytes 
```

so, yeah, we're in a middle of a 128kB page - code is right but traces are misleading as only half right.

Nightwalker-87, 2020-04-20:
Fine, shall we put that into v1.6.1 as well then? Whoever is interested, please assign the ticket to yourself.

chenguokai, 2020-04-21:
I would like to deal with this.
My thought is that as long as the flash page size is varied, additional information can be added like:

`Flash: 0x100000 bytes (1024 KiB) in pages of at least 16384 bytes`

`addr not a multiple of current pagesize (XXX bytes), not supported `

Nightwalker-87, 2020-04-22:
Maybe we should add a second short note like "Check page start address (see RM $cite to doc mentioned by grevaillot (no direct link though)$ )." to give the user a hint on how to proceed. Apart from that the above looks good to me.

chenguokai, 2020-04-22:
It would be indeed better, while it would be more complex, with the need to identify chips and their corresponding RM. If it looks necessary, I will turn my PR as a draft and add more contents.

Nightwalker-87, 2020-04-22:
We shouldn't make it too complicated. Maybe we can also write "Check page start address and compare with flash module organisation in related ST reference manual of your device."
What about that?

chenguokai, 2020-04-22:
> We shouldn't make it too complicated. Maybe we can also write "Check page start address and compare with flash module organisation in related ST reference manual of your device."
What about that?

Added

closed 2020-04-22 by Nightwalker-87