stlink-org/stlink #666: Error build stlink for Fedora Linux

Vascom, 2018-02-14:
I have error build stlink 1.4.0 for upcoming Fedora release.
```
/builddir/build/BUILD/stlink-1.4.0/src/tools/gui/stlink-gui.c: In function 'stlink_gui_set_connected':
/builddir/build/BUILD/stlink-1.4.0/src/tools/gui/stlink-gui.c:517:29: error: cast between incompatible function types from 'void (*)(STlinkGUI *)' {aka 'void (*)(struct _STlinkGUI *)'} to 'void * (*)(void *)' [-Werror=cast-function-type]
     g_thread_new ("devmem", (GThreadFunc) stlink_gui_populate_devmem_view, gui);
```

Full build log https://kojipkgs.fedoraproject.org//work/tasks/2219/25032219/build.log
Other components used for build https://kojipkgs.fedoraproject.org//work/tasks/2219/25032219/root.log

It can be because of GCC 8.0.1.

Please help correct this build.

Vascom, 2018-02-15:
It could be something like this
https://developer.gnome.org/glib/stable/glib-Threads.html#GThreadFunc
```
diff --git a/src/tools/gui/stlink-gui.c b/src/tools/gui/stlink-gui.c
index d45db6f..0af8c2c 100644
--- a/src/tools/gui/stlink-gui.c
+++ b/src/tools/gui/stlink-gui.c
@@ -224,11 +224,14 @@ stlink_gui_update_devmem_view (STlinkGUI *gui)
 
 
 static void
-stlink_gui_populate_devmem_view (STlinkGUI *gui)
+stlink_gui_populate_devmem_view (gpointer data)
 {
     guint            off;
     stm32_addr_t     addr;
 
+    g_return_if_fail (STLINK_IS_GUI (data);
+    STlinkGUI *gui = (STlinkGUI *)data;
+
     g_return_if_fail (gui != NULL);
     g_return_if_fail (gui->sl != NULL);
```

xor-gate, 2018-02-15:
@Vascom could you test latest master?

Vascom, 2018-02-16:
Not work. Same problem https://kojipkgs.fedoraproject.org//work/tasks/5194/25085194/build.log
My code was only example from my friend-programmer.

Vascom, 2018-02-16:
Now #668 looks good and not failed at build.
Are you plan make new release or I should use patch for 1.4.0?

xor-gate, 2018-02-16:
I will create a new release 👍

xor-gate, 2018-02-16:
v1.5.0 has been tagged, enjoy

commit 247b7eb9 (2018-02-15): Try to fix #666 issue (#667)
commit d9a114a6 (2018-02-16): Try to fix 666 issue (#668)

* try to fix #666 issue
* has fixed few function's signatures for match g_thread_new expectations.
* fix check in stlink_gui_populate_filemem_view
* fix gcc-8 warnings
* fix g_return* for proper one
closed 2020-03-18 by Nightwalker-87