stlink-org/stlink #669: Unknown chip id 0042000 on the 2nd st-flash write

Tyrn, 2018-02-18:
- [X] Programmer/board type: Stlink/v2
- [ ] Programmer firmware version:
- [X] Operating system: Linux
- [X] Stlink tools version and/or git commit hash: stlink 1.4.0
- [X] Stlink commandline tool name: e.g `st-flash`
- [X] Target chip (and optional board): STM32F103C8T6

Sorry, being a newbie I've no idea how to check the programmer hardware version.

The following is pretty self-explanatory: first, the blinkyx uploads and blinks; on an attempt at reuploading it bricks. It looks very much like improper usage of st-flash, yet I failed to google a solution. Or the right manual.

Output:

```
$ st-flash write blinkyx.bin 0x8000000
st-flash 1.4.0
2018-02-18T15:12:37 INFO src/common.c: Loading device parameters....
2018-02-18T15:12:37 INFO src/common.c: Device connected is: F1 Medium-density device, id 0x20036410
2018-02-18T15:12:37 INFO src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 1024 bytes
2018-02-18T15:12:37 INFO src/common.c: Attempting to write 8336 (0x2090) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08002000 erased
2018-02-18T15:12:38 INFO src/common.c: Finished erasing 9 pages of 1024 (0x400) bytes
2018-02-18T15:12:38 INFO src/common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2018-02-18T15:12:38 INFO src/flash_loader.c: Successfully loaded flash loader in sram
  9/9 pages written
2018-02-18T15:12:38 INFO src/common.c: Starting verification of write complete
2018-02-18T15:12:38 INFO src/common.c: Flash written and verified! jolly good!

$ st-flash write blinkyx.bin 0x8000000
st-flash 1.4.0
2018-02-18T15:19:56 INFO src/common.c: Loading device parameters....
2018-02-18T15:19:56 WARN src/common.c: unknown chip id! 0xe0042000

$ st-flash erase
st-flash 1.4.0
2018-02-18T15:20:25 INFO src/common.c: Loading device parameters....
2018-02-18T15:20:25 WARN src/common.c: unknown chip id! 0xe0042000
```

Thank you,
The stlink project maintainers

lhoyosm, 2018-09-13:
I've got a very similar issue with almost the same conditions, I got instead:

```
$st-flash --format ihex write out.hex
st-flash 1.5.0
2018-09-12T20:47:17 INFO common.c: Loading device parameters....
2018-09-12T20:47:17 WARN common.c: Invalid flash type, please check device declaration
Unknown memory region
```
I could fix that (de-brick it) by moving the boot jumpers, power cycling the board and reprogramming it with the arduino ide, then it was OK again

It seems that I am doing something wrong with the linker script or the st-flash
Hope it helps
Additional help is appreciated

Nightwalker-87, 2020-02-21:
Feedback closes #669.

Nightwalker-87, 2020-03-19:
@Tyrn @lhoyosm: What is the current status of this issue? Is it hardware- or software-related?
Please test with Release v1.4.0 and v1.5.0 first.

lhoyosm, 2020-03-19:
For me the issue was due to a wrong configuration of the STM32 Chip that caused the communication to fail. After programming, It looked like it was "bricked" as the SWDIO and SWCLK pins were reconfigured as GPIO and the error message was displayed when trying a reprogram.

I think "unknown chip id! 0x_______" is a synonym of a connection problem, if nothing is connected that is displayed. Maybe a message can be added regarding that to check connections.

The st-flash was not the issue. Just tested 1.4.0, 1.5.0 and 1.6.0 all working fine!

Nightwalker-87, 2020-03-19:
Thank you very much for the feedback which appears to be helpful.
The idea of improving the displayed message does make sense to me.
I'll check this and forward it as a feature request to Release v1.6.1.

Nightwalker-87, 2020-07-18:
The meaning of the output `unknown chip id! 0xe0042000` should be documented, as well as some other error codes.

Nightwalker-87, 2021-03-12:
Duplicate of #568.

commit 229c7211 (2021-03-24): General Project Update

- Added basic security policy
- Updated info on version support for macOS
- Updated tutorial.md
> Added reference for st-info --probe command
> Added section on unknown chip id error
   (Closes #107, Closes #568, Closes #669)
> Added information on HW reset pin issue
   (Closes  #220, Closes #238)
closed 2021-03-24 by Nightwalker-87